Koenkk/zigbee-herdsman #789: How to deal with same manufcaturer/cluster combo but different attributes?

sjorge, 2023-10-27:
Interesting problem I am hitting,

For the Ubisys H10 we added manuSpecificUbisysHeatingRegulatorThermostat which is the same ID as hvacThermostat but with the ubisys manufacturer code, it has a bunch of H10 specific attributes.

For the Ubisys H1 they just introduce setting vacation mode, it's also on the hvacThermostat cluster ID with ubisys manufacturer code. 

```
The ID of the manufacturer-specific attribute vacation mode is 0x0005 with the manufacturer ID being 0x10f2
```

Because they both use H10 and H1 use hvacThermostat + ubisys manufc code for custom attribute there is a collision:

https://github.com/Koenkk/zigbee-herdsman/blob/e54db7a20bb442d8be7bef2de837772cdcbfaec3/src/zcl/definition/cluster.ts#L4273C1-L4273C66

Adding a 2nd one with the attribute for the H1 causes issues in zh, also just adding a duplicate attribute obviously doesn't work either.

How do we resolve this cleanly?

sjorge, 2023-10-27:
Found a workaround for now, as I only need to write and not read, I can just do manual write datatype all provided.

Would be nice to have a way to handle this cleanly in the future somehow though. I guess adding a unique ID to the cluster definition and somehow having the zhc device definition say use this one for me would work.

Koenkk, 2023-10-27:
Since this is the first time we encounter such thing and we can get around it with the manual read/write I propose to not do anything for now (it will have quite some impact on the code base I think).

sjorge, 2023-10-27:
Can we start a know limitstion list somewhere to track these things? I find it hard to believe I'm the first to hit this.

Koenkk, 2023-10-27:
Yes, I've put the `dont-stale` label and will link all similar issues here.

closed 2024-05-05 by Nerivec