Koenkk/zigbee-herdsman #803: Cannot read statuses and control devices. Adapter: ezsp

JakubGoliasz, 2023-11-17:
### What happened?

I cannot control Zigbee devices or read their status. There is also a crash at the end.

*Adapter: ezsp*

### What did you expect to happen?

Ability to receive messages from devices and control them.

### How to reproduce it (minimal and precise)

Just run the Zigbee2MQTT 1.33.2 add-on in HA. Rollback to 1.33.1-1 helps.

*Adapter: ezsp*

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle

### Debug log

```
Zigbee2MQTT:debug 2023-11-02 07:31:08: Publishing 'set' 'state' to 'Lampa'
Zigbee2MQTT:error 2023-11-02 07:31:08: Publish 'set' 'state' to 'Lampa' failed: 'Error: Command 0x04cf8cdf3c7e4d4c/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Zigbee2MQTT:debug 2023-11-02 07:31:08: Error: Command 0x04cf8cdf3c7e4d4c/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:471:19)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

```
Zigbee2MQTT:info  2023-11-02 07:42:07: MQTT publish: topic 'zigbee2mqtt/Czujnik zalania', payload '{"battery":100,"battery_low":false,"device_temperature":27,"linkquality":128,"power_outage_count":12,"tamper":false,"voltage":3115,"water_leak":false}'
Error: ZdoRequest error
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:572:19)
    at EZSPAdapter.nodeDescriptorInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:380:28)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:370:32)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

Farmanden, 2023-11-21:
Got similar issue here, mostly timeout on a queue, I will report the next error io get. WAF is low when the lights are not working in the morning :sweat_smile:.

Farmanden, 2023-11-22:
Here we go, crash back trace: 

Zigbee2MQTT:info  2023-11-22 06:08:55: MQTT publish: topic 'zigbee2mqtt/0x040d84fffed5f087', payload '{"battery":66,"battery_low":true,"elapsed":120275,"linkquality":184,"local_temperature":20,"occupied_heating_setpoint":20,"pi_heating_demand":4,"running_state":null,"voltage":2600}'
Zigbee2MQTT:info  2023-11-22 06:09:28: MQTT publish: topic 'zigbee2mqtt/PenthouseBadRadiator', payload '{"ADC":"1375,1370,1,224,0,0,0,0,0,226,228,237","battery":66,"battery_low":true,"elapsed":60360,"linkquality":124,"local_temperature":22.4,"occupied_heating_setpoint":0,"pi_heating_demand":0,"running_state":null,"voltage":2600}'
Error: Connection not initialized
    at Ezsp.execCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at EZSPAdapterBackup.createBackup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/backup.ts:21:56)
    at EZSPAdapter.backup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:618:31)
    at Controller.backup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:328:47)

closed 2024-05-05 by Nerivec