Koenkk/zigbee-herdsman #834: Unhandled EZSP frame incomingManyToOneRouteRequestHandler can cause EZSP to stop

irata-coder, 2023-12-21:
### **What happened?**
Unhandled EZSP frame `incomingManyToOneRouteRequestHandler` while performing an OTA update of another device, causes EZSP to stop

### **What did you expect to happen?**
Receipt of this frame should not cause EZSP and driver to stop

### **How to reproduce it (minimal and precise)**
1.) Create a Zigbee network with z2m and sonoff ZBDongle-E with 7.3.2.0 build 212 (albeit also recreatable on earlier builds)
2.) Add Hive 1613V Active Plug and Bosch BTH-RA
3.) Start an OTA update on Bosch BTH-RA
4.) Within a few minutes, this frame will arrive and the error is seen, stopping the driver

### **Zigbee2MQTT version**
1.34.0

### **Adapter firmware version**
 7.3.2.0 build 212 

### **Adapter**
EZSP Sonoff Dongle Plus-E EFR32MG21

### **Debug log**

```
  zigbee-herdsman:adapter:ezsp:uart <-- [6273b1a9292ab455fcab6527a30b8c499824c77e] +393ms
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,2,0): 6273b1a9292ab455fcab6527a30b8c499824c77e +392ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (7) +0ms
  zigbee-herdsman:adapter:ezsp:uart --> [87009f7e] +393ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3190017d00a1e7a53f2f02095e1e0004 +393ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x7d: {"_cls_":"incomingManyToOneRouteRequestHandler","_id_":125,"_isRequest_":false,"source":59297,"longId":{"type":"Buffer","data":[0,30,94,9,2,47,63,165]},"cost":4} +1ms
  zigbee-herdsman:adapter:ezsp:driv Unhandled frame incomingManyToOneRouteRequestHandler +2s
  zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":3} after 2000ms +21ms
  zigbee-herdsman:adapter:ezsp:uart -!- break rewaiting (3) +0ms
  zigbee-herdsman:adapter:ezsp:uart Can't resend DATA frame (2,0,1): 3200017e002672f1feffdbb48401 +0ms
  zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 0 +19ms
  zigbee-herdsman:adapter:ezsp:driv Stop driver +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp +20ms
```

Koenkk, 2023-12-30:
@kirovilya could you check this?

Bascht74, 2024-02-06:
@kirovilya Could you take a look at that?

Here are some logs of my adapter: https://github.com/Koenkk/zigbee2mqtt/discussions/21011

closed 2024-09-04 by github-actions[bot]