Koenkk/zigbee-herdsman #847: endpoind.read() should also use attribute specific manufacturerCode when available

sjorge, 2023-12-27:
Related to #843 we should also just use the information when doing a read.

However this does not seems as straight forward as with the reporting configuration.

Since `endpoint.read()` can take a array of attributes. As long as they have the same manufacturerCode requirement this is not an issue but in theory it could be a mixed case! For example one could read `localTemp` + `ubisysRemoteTemperature` from `hvacThermostat` 

https://github.com/Koenkk/zigbee-herdsman/blob/5411ab6ce77e218b076628337e7001c59914f3d5/src/zcl/definition/cluster.ts#L1974

https://github.com/Koenkk/zigbee-herdsman/blob/5411ab6ce77e218b076628337e7001c59914f3d5/src/zcl/definition/cluster.ts#L2085

https://github.com/Koenkk/zigbee-herdsman/blob/5411ab6ce77e218b076628337e7001c59914f3d5/src/controller/model/endpoint.ts#L445-L447

The information would be available in the loop above in the `cluster.getAttribute()` result.

We would need to split this into N frames:

https://github.com/Koenkk/zigbee-herdsman/blob/5411ab6ce77e218b076628337e7001c59914f3d5/src/controller/model/endpoint.ts#L449-L453 would need to be called one time for each manufacturerCode (null also being one if an attribute has none)

We would also need to collect the responses:
https://github.com/Koenkk/zigbee-herdsman/blob/5411ab6ce77e218b076628337e7001c59914f3d5/src/controller/model/endpoint.ts#L462-L465

I think before we have not hit this as we seem to always provide the manufacturerCode ourselves, but there should be no reason to do so as we have the information.

A alternative and simpler solution would be to lookup the manufacturerCode here:
https://github.com/Koenkk/zigbee-herdsman/blob/5411ab6ce77e218b076628337e7001c59914f3d5/src/controller/model/endpoint.ts#L445-L447

But only update options.manufacturerCode if it is currently undefined, if it is already defined we check if the one we got matched the one in options and we throw and error if they do not match.

It would then be up to the called to split the `endpoint.read()` so instead of reading both `localTemp` + `ubisysRemoteTemperature` , one would read `localTemp` and then `ubisysRemoteTemperature`.

@Koenkk I can manage to do the alternative described at the end, that should not be too hard. However I am not certain on the in my opinion better approach. The alternative would fix my problem as I won't be doing a mixed read of attributes with different manufacturerCode (well null + with a code). I think throwing an error if someone does try to do a mixed read at some point is acceptable. If a lot of people run into it we can always come back and handle it better.

(I think this will also help people doing manual reads over mqtt, as a common mistake seems to be forgetting to pass in the manufacturerCode)

sjorge, 2023-12-27:
After giving it some more though in the shower, I think the simpler alternative described at the end would be sufficient and a lot less complicated.

Koenkk, 2023-12-28:
>  I think throwing an error if someone does try to do a mixed read at some point is acceptable

Yes, was also thinking about this, it's fine.

sjorge, 2023-12-28:
As per our chat:
- Same approach for `endpoint.write()`
- Second pass over `endpoint.configureReporting()` as it also takes multiple reports.

sjorge, 2023-12-28:
I've hit a first snag :(

Let's take the hvacThermostat.localTemp + hvacThermostat.ubisysRemoteTemperature as an example of 2 attributes to read/write/configure in one go. (I was going to fix up endpoint.configureReporting() first)

## Situation
- hvacThermostat: no cluster level manufacturerCode (cluster.manufacturerCode == null)
- hvacThermostat.localTemp: has ID, type but no manufacturerCode returned in the cluster.getAttribute() call.
- hvacThermostat.ubisysRemoteTemperature: has ID, type, and manufacturerCode returned in property in the cluster.getAttribute() call.

## Code paths
### localTemp before ubisysRemoteTemperature
1. options.manufacturerCode get set to null (coming from cluster)
2. first time through the map() -> we do nothing as localTemp has no manufacturerCode property returned
3. second time through the map() -> we update options.manufacturerCode to 4338, we allow this as options.manufacturerCode  was unset (well null in this case)

We now have a frame that has 2 attribute with different manufactuereCode requirements, oops! We can't detect as options.manufacturerCode  is never undefined it's null or a number. So we don't know if it's null because it's the cluster default or there as an attribute before us that didn't need a manufacturerCode.

### ubisysRemoteTemperature before localTemp
1. options.manufacturerCode get set to null (coming from cluster)
2. first time through the map() -> we update options.manufacturerCode to 4338
3. second through the map() -> we do nothing as localTemp has no manufacturerCode property returned

We could always return manufacturerCode for getAttribute() set to null , then localTemp will have a manufacturerCode property and we can detect this when we try to update  options.manufacturerCode from 4338 back to null which is not allowed so we throw the exception. This however does nothing if localTemp is before ubisysRemoteTemperature in iterations through the map.

sjorge, 2023-12-28:
Actually, we should probably have `cluster.getAttribute()` always return a `manufacturerCode` set to null.

^ @Koenkk will that break things ?

If not, we can reduce or was it map I always confuse those, our set of attributes and if we get more than 1 manufacturerCode out of it, throw the exception in read/write/configureReporting ?

sjorge, 2023-12-28:
Existing test are a life saver! Initial pass had a few interesting regressions.

closed 2024-01-05 by sjorge