Koenkk/zigbee-herdsman #854: zigbee2mqtt<->EZSP incompatible with latest Gecko SDK v4.4.0.0 (EmberZNet 7.4.0.0) firmware build

xconverge, 2024-01-05:
Updated to the latest silabs firmware (yes I know this is beta but figured I would at least track it here....)

Startup fails:
```
Zigbee2MQTT:info  2024-01-04 19:45:19: Logging to console only'
Zigbee2MQTT:info  2024-01-04 19:45:19: Starting Zigbee2MQTT version 1.35.0 (commit #b5a90e9)
Zigbee2MQTT:info  2024-01-04 19:45:19: Starting zigbee-herdsman (0.30.0)
Zigbee2MQTT:debug 2024-01-04 19:45:19: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Error: {"address":0,"clusterId":32770,"sequence":2} after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
Using '/app/data' as data directory

```
rolling back to 7.3.2.0 runs great

I suspect this _could_ be the relevant change?
https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.4.0.0.pdf
```
  Removed in release 7.4.0.0
      ...
  • Removed setPacketBufferCount() in af-host.c and useless check case EZSP_CONFIG_PACKET_BUFFER_COUNT: in command-
  handlers.c.
     ...
```

And this is where it is used
https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/ezsp/driver/ezsp.ts#L487

xconverge, 2024-01-05:
@kirovilya I don't have a dev environment setup to check into this too much. I am unsure how to run test changes locally/easily

I wanted to add some logging to see exactly which config failed to set to confirm

kirovilya, 2024-01-05:
where can I get ncp firmware 7.4.0.0?

xconverge, 2024-01-05:
> where can I get ncp firmware 7.4.0.0?

I have a skyconnect so got it here:

https://github.com/NabuCasa/silabs-firmware

I am unsure if you have an adapter with a different pin out or is incompatible with these builds unfortunately

wastez, 2024-01-05:
Or here for other adapters (it's a fork of the silabs source)

https://github.com/darkxst/silabs-firmware-builder

NewsGuyTor, 2024-01-06:
This is actually relatively urgent, this issue broke a lot of Zigbee2MQTT installations today. More info here:
https://github.com/Koenkk/zigbee2mqtt/issues/20602

xconverge, 2024-01-06:
> This is actually pretty urgent, this issue [broke a lot of Zigbee2MQTT installations today,](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/555) and will break more the next time "Silicon Labs Multiprotocol" is updated for HA users.
> 
> More info here: [Koenkk/zigbee2mqtt#20602](https://github.com/Koenkk/zigbee2mqtt/issues/20602)

I am surprised the addon pushed a FW update so quickly, I thought I was on the cutting edge here (and I still kind of think I am) and was reporting/looking into it a bit just to be polite/allow developers to get ahead of it (Especially since a lot historically falls onto @kirovilya at least in this project (much appreciated by the way!))

kirovilya, 2024-01-06:
[image]
Some commands that were previously used were removed there. so I have to rewrite the code... not quickly

kirovilya, 2024-01-09:
See https://github.com/Koenkk/zigbee-herdsman/issues/854, should be fixed in the dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

burntoc, 2024-01-09:
I haven't seen the updated release in the dev branch yet, just the commits and testing activity.

wastez, 2024-01-09:
Have seen the Update is available in z2m edge.
Has anybody tried it?

burntoc, 2024-01-09:
Working here.

chris-1243, 2024-01-10:
@kirovilya 

I just had a try on the new SDK 4.4.0 and the dev branch of z2m and I had those warnings: 

01/10/2024 04:17:14 PM Using '/app/data' as data directory
01/10/2024 04:17:17 PM Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
01/10/2024 04:17:17 PM Assertion failed: exportKey returned unexpected status: 45

Z2m is working. I revert to SDK 4.3.0 (firmware 7.3.0.0, SONOFF DONGLE-E) on which I have no warnings. SDK 4.3.1 and higher provide me this warning only:

Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]

Zigbee herdmann below for some debugging if it may help (SDK 4.4.0, firmware 7.4.0.0)

[log_herdsman.txt](https://github.com/Koenkk/zigbee-herdsman/files/13890145/log_herdsman.txt)

kirovilya, 2024-01-10:
@chris-1243 thanks for checking. yes, there are some warnings, but they not critical... I’ll continue to improve it as soon as I have free time

chris-1243, 2024-01-10:
@kirovilya

Great news. I am going to re-flash my coordinator with the latest SDK and see how it will perform

joe-sydney, 2024-01-13:
Just swapped to the Edge Z2M addon on HASS, updated the Silabs Multipan add to latest (with the new Gecko SDK) and also flashed the Sonoff Dongle E to 4.4.0.... all working so far!

Nerivec, 2024-01-28:
Anyone in here that has tested EDGE for a decent length of time, could you drop a line [here](https://github.com/Koenkk/zigbee2mqtt/discussions/21011), so we know what errors some people might still be seeing (or if it works great 😄)? Thanks.

xconverge, 2024-02-02:
Issue is fixed now that 1.35.2 is released. I am on 7.4.0.0 and will stick it out to see if anything arises, thanks @Nerivec and @kirovilya

closed 2024-02-02 by xconverge

newlund, 2024-02-11:
> Issue is fixed now that 1.35.2 is released. I am on 7.4.0.0 and will stick it out to see if anything arises, thanks @Nerivec and @kirovilya

Is 7.4.0.0 working for you? Thinking of upgrading from 7.3.2.0 :)

chris-1243, 2024-02-11:
7.4.0.0 is working on version 1.35.2 or higher. 

 Just be carefull. RCP firmware based on SDK 4.4.0 (multi-protocol with silabs addon) is problematic and has been downgraded to SDK 4.3.1.

NCP firmware 7.4.0.0 (Zigbee only) does not seem to be affected by any issues and is working fine. Have a look to this [link](https://github.com/darkxst/silabs-firmware-builder/issues/48)

newlund, 2024-02-11:
Thanks! I only use zigbee for now and run the NCP firmware so guess I could upgrade. But think I will wait a while... Latest is not always greatest... :)

johanymerson, 2024-02-16:
Just tested 1.35.3 with a Sonoff ZBDongle-E running NCP 7.4.0.0. It died within 30 minutes with the following error:
Error: {"address":64843,"clusterId":32801,"sequence":30} after 10000ms
    at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at Driver.zdoRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:667:43)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:566:13
    at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)