Koenkk/zigbee-herdsman #910: Z2M 1.35.2 stops with "Adapter disconnected, stopping" after few minutes of uptime (Sonoff-E / EZSP v12 / FW 7.3.2.0 build 212)

romarysonrier, 2024-02-05:
**1./ A first, i was running Z2M 1.35.1 withon a 2 device setup:**
- SONOFF	SNZB-02D
- LiXee	ZLinky_TIC
Adapter is SONOFF-E / EFR32 MGM21 (upgraded to EZSP v12 / FW 7.3.2.0 build 212)

The  ZLinky_TIC device is using a lot of polling to report some measurement related to my Electric Meter (Linky).
The most simple way to monitor how well the polling is working is to track how often is update a TIMEDATE valiue

While running Z2M 1.35.1, i was experiencing some unexpected random freeze any polling activities from Z2M, after some hours of uptime. Z2M never stopped or crashed. Only data from Reportable attributes were still updated. Network scan was only stuck and not responding once the polling issue was observed.

When working the polling did offers very low refresh rate for ZLinky_TIC, compared to the 60 sec requested ,  TIMEDATE from Electric Meter was updated every 70sec to 12 min, average was near 2 min. And at least a day the whole netflow was suffering from lost of all polled attributes.

I struggled to identify a clear root cause at this point, could also be hardware since i am using DIY case.

**2./ But, after upgrading  Z2M 1.35.2, the 'abnormal' behaviour changed:**
- refresh rate for ZLinky_TIC is almost always 70sec (good progress)
- Z2M stops working randomly after some minutes of uptime ("Adapter disconnected, stopping" , no suspicious log) and restart automaticaly 

```
info  2024-02-05 14:15:36: MQTT publish: topic 'zigbee2mqtt-grenier/ENEDIS Linky Portail', payload '{...} ,"update_available":null}'
debug 2024-02-05 14:16:01: Received Zigbee message from 'ENEDIS Linky Portail', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"apparentPower":217}' from endpoint 1 with groupID 0
info  2024-02-05 14:16:02: MQTT publish: topic 'zigbee2mqtt-grenier/ENEDIS Linky Portail', payload '{...} ,"update_available":null}'
error 2024-02-05 14:16:09: Adapter disconnected, stopping
debug 2024-02-05 14:16:09: Saving state to file /opt/zigbee2mqtt/data/state.json
info  2024-02-05 14:16:09: MQTT publish: topic 'zigbee2mqtt-grenier/bridge/state', payload '{"state":"offline"}'
info  2024-02-05 14:16:09: Disconnecting from MQTT server
info  2024-02-05 14:16:09: Stopping zigbee-herdsman...
error 2024-02-05 14:16:09: Failed to stop Zigbee2MQTT
```

One hypothesis to explain the abnormal behavior is that the intesive use of polling done by the ZLinky_TIC is causing some issue with protocol  EZSP v12 implementation.

Nerivec, 2024-02-05:
https://github.com/Koenkk/zigbee2mqtt/issues/21198

After some investigation in that issue, it seems yours might be the same. The presence of the LiXee devices in both reports suggests something...

I'll continue answering in that issue if you don't mind joining there, to keep things organized (unless we eventually find different problems).
If you have a [herdsman debug log](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging) of the crash, and can post it there, that'd be great.

romarysonrier, 2024-02-05:
here is the debug report
[lixee-2.log](https://github.com/Koenkk/zigbee-herdsman/files/14170047/lixee-2.log)

closed 2024-05-04 by Nerivec