Koenkk/zigbee-herdsman #933: Load additional manufacturer-specific clusters from device converter

rgiordano30, 2024-02-25:
Hi, with new zigbee chips like Espressif ESP32-C6, more and more personal/hobby projects using the Zigbee protocol are starting to appear.

Problem is that manufacturer-specific clusters (and attributes) are hardcoded in zigbee-herdsman and there are no ways to extend them without modifing the source code.

It's would be great to add a new "cluster" option to the device converter and load it in zigbee-herdsman when parsing messages.

sjorge, 2024-03-05:
Been wanting this for a while too, it would be great to move those non ZCL ones to the device definitions.

@koenkk I wonder if can somehow leverage some of the extend stuff to inject (and also share some between devices)

Although thinking about it some more that won't work as all the code to resolve these lives in zh. 🤔

Koenkk, 2024-03-05:
@sjorge it needs to be injected somehow into ZH during startup, can be done in `onEvent` but didn't had the time to look at this yet.

Suxsem, 2024-03-07:
@Koenkk  Yeah I'm really waiting for this one too, it would open a world of possibilities for makers!
If you are willing to provide me some insights in how you would implement this I'm happy to try making a PR!

Suxsem, 2024-03-11:
@Koenkk 
I see that in https://github.com/Koenkk/zigbee2mqtt/blob/2a53b8e59b457e0805318bdb38d2c74de651724d/lib/zigbee.ts#L61
ZH is initialized with an herdsmanSettings object, do you think this could be a good place to inject manufacturer-specific clusters?
Or this settings should only contain global (aka adapter) settings?

closed 2024-05-05 by Nerivec