Koenkk/zigbee-herdsman #940: [Proposal] Removal of "forever" Permit Join

Nerivec, 2024-02-27:
After observing behaviors while testing `ember` and seeing reports here and there about this, I think removing the possibility to open the network "forever" via internal workarounds would make sense.

1) It doesn't seem to work well in general from Discord reports (compared to triggering as needed with the Permit Join button).
2) Some stacks now enforce no "forever", which messes with the interval refresh triggered by Z2M, basically breaking the "assumed state".
3) It prevents state validation, because of the interval refresh that won't actually do anything if the network is already open (at least in EmberZNet)
4) It's bad for security...

The TODOs on this would appear to be:
- Remove the config 
- Cleanup the config validations
- Remove the interval refresh logic to only have:
  - Open for 254 seconds (_0xFF can be a special value_) on button click.
  - Close on button click if within 254 seconds of opening.
  - Optionally use stack-level function/prop to determine if network is currently opened instead of assuming or using an internal timer (not sure if all stacks support this)
- Remove/fix related tests

Related code (probably not all...):

https://github.com/Koenkk/zigbee-herdsman/blob/b004cf827da04eb78e7658776bd2726289797b16/src/controller/controller.ts#L237

https://github.com/Koenkk/zigbee2mqtt/blob/f04ade6d764518bd8228eb32a15e96cd4e3fbde6/data/configuration.yaml#L5
https://github.com/Koenkk/zigbee2mqtt/blob/f04ade6d764518bd8228eb32a15e96cd4e3fbde6/lib/types/types.d.ts#L122
https://github.com/Koenkk/zigbee2mqtt/blob/f04ade6d764518bd8228eb32a15e96cd4e3fbde6/lib/util/settings.schema.json#L48
https://github.com/Koenkk/zigbee2mqtt/blob/f04ade6d764518bd8228eb32a15e96cd4e3fbde6/lib/util/settings.ts#L39
https://github.com/Koenkk/zigbee2mqtt/blob/f04ade6d764518bd8228eb32a15e96cd4e3fbde6/lib/controller.ts#L139
https://github.com/Koenkk/zigbee2mqtt/blob/f04ade6d764518bd8228eb32a15e96cd4e3fbde6/lib/extension/bridge.ts#L161

Koenkk, 2024-02-27:
There are some special cases where permit join is required to be permanently enabled, if I remember correctly for some green power devices. As a simple first step, let's just remove the `permit_join` option from the Z2M `configuration.yaml`. Then we can see if any setups break.

Nerivec, 2024-02-27:
Just putting `false` as override would be a good first step for sure (just ignore the config at first). Better off waiting until after the 1st to implement any of this; just thought I'd get the ball rolling on the topic 😉 

> There are some special cases where permit join is required to be permanently enabled, if I remember correctly for some green power devices.

Is it the network that needs to be open, or the [commissioning mode](https://github.com/Koenkk/zigbee-herdsman/blob/b004cf827da04eb78e7658776bd2726289797b16/src/controller/greenPower.ts#L225) that needs refreshing?
I guess that refresh logic could be moved to GP if it's the second case (I suppose a separate button would be ideal in that scenario, kind of like Touchlink... make it clear what's what).
@dduransseau @chris-1243 Sorry for the ping, again, but I guess you've gained the GP-go-to status 😁

chris-1243, 2024-02-27:
For the PTM215Z/216Z, there is no need to have the `permit join` always on, on my side at least. I do open the network for the pairing and close it when done.

In `configuration.yaml` permit join is set to `false` for me.

Nerivec, 2024-02-27:
@chris-1243 Can you confirm this is the case in zStack too? I believe you have access to a network using it.

@dduransseau If that's confirmed, my guess would be that the coordinator is always in commissioning mode for GP (no matter what Z2M says), but that would likely be EmberZNet-specific, and coordinator-only.

dduransseau, 2024-02-28:
Nope, I figure out this behavior on deconz with conbee II for the first time. I'm not an expert of herdsman, since GP provisioning do not use same type of frame, would it be possible that GP frame bypass the allow_join filter ?

chris-1243, 2024-02-28:
@Nerivec I need to check if this behaviour occurs between my main network (zstack) and test (ember).

As I have both PTM215/216, I will try with both of them. As far as I remember, I have not seen such behaviour.

My two networks are not on the same channel and do not share the same network keys, pan id and extended id. All those values are different.

As you have to choose your network channel when pairing, it sounds strange for me to see the same device connected to two differents networks.

dduransseau, 2024-02-28:
In my scenario both network are on the same channel but different network and pan key.

dduransseau, 2024-02-29:
I just found [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/19848) that seems related

MattWestb, 2024-03-03:
> In my scenario both network are on the same channel but different network and pan key.

Make sure also have different Extended Pan-ID or you can getting strange problems then paring and network managements commands in both networks.

Hedda, 2024-09-12:
> After observing behaviors while testing `ember` and seeing reports here and there about this, I think removing the possibility to open the network "forever" via internal workarounds would make [...]

Still makes sense to me.

closed 2024-10-11 by Nerivec