Koenkk/zigbee-herdsman #954: Ikea Motion Sensor E1525/E1745 not updating properly

Nerivec, 2024-03-06:
https://www.zigbee2mqtt.io/devices/E1525_E1745.html

The device is not properly updating its state. Some users report only occupancy not updating, some users report all properties not updating.
The issue is (or has been) present in [other drivers](https://github.com/Koenkk/zigbee2mqtt/discussions/21462#discussioncomment-8634218) (even [other software than Z2M](https://community.smartthings.com/t/no-motion-detected-from-ikea-tradfri-motionsensor/249410/40)), but the `ember` driver currently seems to make it a "persistent issue".

[Latest logs](https://github.com/Koenkk/zigbee2mqtt/files/14502727/z2m-ikea.log) (@wastez)

@Balooforever @dspalu32 @wastez

[Original thread](https://github.com/Koenkk/zigbee2mqtt/discussions/21462#discussioncomment-8627019)

Nerivec, 2024-03-06:
Can one of you contact me on Z2M Discord when you have some time to run a few tests please?

wastez, 2024-03-06:
Yes sure, give me a hour and then I’m ready to test.

wastez, 2024-03-06:
Is that ok for you? How can i contact you?

Nerivec, 2024-03-06:
Sure. Ping me @Nerivec in Z2M Discord.

closed 2024-03-08 by Koenkk

wastez, 2024-03-09:
Really working good now, also the problem after the restart seems to be gone.

Great!!!

Nerivec, 2024-03-09:
Thanks for the help testing this @wastez 😉

PR #959 (2024-03-08): [ember] Reworked multicast registration on coordinator
According to ZCL cluster spec, this should take care of state reporting issues encountered with `onOff` "global scene" usage by certain devices.

Fixes #954

_Cleanup some comments._