Koenkk/zigbee-herdsman #979: Move special readResponse out of zhc/src/index.ts (fixes legrand pairing issues)

sjorge, 2024-03-16:
Currently there is some extra logic in index.ts in zhc repo to handle pairing of legrand devices and some quirks for aqara devices.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/523c1e7147a2178382c9cb883e047bffdd0c2a18/src/index.ts#L351-L374

Somewhere after around z2m 1.34.0 we lost the ability to pair legrand devices because zh is also sending readResponses here https://github.com/Koenkk/zigbee-herdsman/blob/e3b17c2b2c833b5b80f7b4d70394fdfc627094d7/src/controller/model/device.ts#L259

Which result in the legrand device immediately leaving the network, we do send the response from zhc/src/index.ts after but at that point it's already to late.

I had a quick look at this earlier but it seems the data available there is still not completely parsed and I was not able to identify for which device the read was coming from or if it included the marker attribute legrand uses. (We should probably also add that to the cluster definition at some point).

Or perhaps an opt-out for these responses on a per-device level, so those can stay in zhc.

Koenkk, 2024-03-17:
Can you check if this issue is fixed by applying the following changes?

- https://github.com/Koenkk/zigbee-herdsman-converters/pull/7228
- https://github.com/Koenkk/zigbee-herdsman/pull/982

sjorge, 2024-03-17:
This seems to work!

I removed the legrand din meter, then, enabled pairing and initiated pairing on the device again. It successfully reached the configure stage and the device is not leaving the network!

sjorge, 2024-03-17:
Hmm I may have spoken to so, the device did not leave but it does appear it's not reporting anything.

sjorge, 2024-03-17:
OK, got it to show the green led.

It seems I had to stop and start z2m between removing it (device left when request) and starting the pairing on the device. Perhaps some state left over somehow ?

Koenkk, 2024-03-17:
> It seems I had to stop and start z2m between removing it (device left when request) and starting the pairing on the device. Perhaps some state left over somehow ?

Yes, after removing a device you need to restart z2m to clear the whole state.

Koenkk, 2024-03-19:
@sjorge can you confirm that everything works correctly in the latest dev branch?

closed 2024-03-19 by Koenkk

sjorge, 2024-03-20:
Still working 👍

Pyla5-Katy0-Vibe7, 2024-03-26:
I also confirm that the :dev branch is required to avoid loop pairing. It works well.

r0m1CH, 2024-03-26:
I confirm we need :dev branch in order to fix loop legrand pairing issue. Thanks.