Koenkk/zigbee-herdsman #1037: State of ZiGate support

Koenkk, 2024-04-28:
In https://github.com/Koenkk/zigbee-herdsman/pull/1033 some issues with the ZiGate adapter were discovered, some parts of the code look broken. Since I've not seen any issues reported about ZiGate, I'm wondering wether anyone is still using it and if it still works.

Tagging some people who have worked on or used the ZiGate adapter: @G1K @devbis @kirovilya @yateeshc @eswenson1 @incognito1234 @rclsilver @darkxst @JC38 

**My question to you**: are you still using the ZiGate adapter and if yes, does it work fine?

eswenson1, 2024-04-28:
I’m using Zigate with ThirdReality Zigbee USB dongle. After updating the firmware that came with the dongle, with works fine with HA OS/Zigbee2MQTT.

Nerivec, 2024-04-29:
Some of the issues identified (that I could remember), related to frames handling only (may be others in other areas):

Inconsistent position increase:
https://github.com/Koenkk/zigbee-herdsman/blob/e90595a0b57d2f8e185744c7e64bf3f824d4f7c7/src/adapter/zigate/driver/buffaloZiGate.ts#L14

Typo resulting in passthrough to `Buffalo` (would also `read` instead of `write`):
https://github.com/Koenkk/zigbee-herdsman/blob/e90595a0b57d2f8e185744c7e64bf3f824d4f7c7/src/adapter/zigate/driver/buffaloZiGate.ts#L21

Typo resulting in passthrough to `Buffalo`:
https://github.com/Koenkk/zigbee-herdsman/blob/e90595a0b57d2f8e185744c7e64bf3f824d4f7c7/src/adapter/zigate/driver/buffaloZiGate.ts#L77

Inconsistent read/write (and inconsistent usage):
https://github.com/Koenkk/zigbee-herdsman/blob/e90595a0b57d2f8e185744c7e64bf3f824d4f7c7/src/adapter/zigate/driver/buffaloZiGate.ts#L103
https://github.com/Koenkk/zigbee-herdsman/blob/e90595a0b57d2f8e185744c7e64bf3f824d4f7c7/src/adapter/zigate/driver/buffaloZiGate.ts#L108

Broken (`messageType.ts` and `commandType.ts` both need a detailed look-over):
https://github.com/Koenkk/zigbee-herdsman/blob/e90595a0b57d2f8e185744c7e64bf3f824d4f7c7/src/adapter/zigate/driver/messageType.ts#L215

Missing types (enum isn't used at all...):
`BUFFER_RAW, UINT16BE, UINT32BE, LIST_UINT16BE, MAYBE_UINT8, LOG_LEVEL, STRING`
https://github.com/Koenkk/zigbee-herdsman/blob/e90595a0b57d2f8e185744c7e64bf3f824d4f7c7/src/adapter/zigate/driver/parameterType.ts#L1

---

_Some issues were corrected (hopefully) in https://github.com/Koenkk/zigbee-herdsman/pull/1033 but some stuff needs to be looked over by someone with zigate-specific knowledge._

JC38, 2024-04-29:
Hello,
I am still using it, but just with 8 devices that I need to migrate to my other controllers (I have 4 zigbee2mqtt ;-) ) 
No issues with the 8.

darkxst, 2024-04-30:
As mentioned above the Third Reality Zigbee Dongle, which is a relatively new product is using Zigate backend. I tried to ping a contact I have there about this, however his email is bouncing. However they are part of the "works with Home Assistant" program and so I would guess they make some efforts to support this backend.

eswenson1, 2024-04-30:
I have a contact at ThirdReality who has been very responsible and helped me get things working. I hesitate to put his email address here, but it you want to contact me directly, I’ll give you his email.

Nerivec, 2024-04-30:
@eswenson1 If you can contact that person yourself, and let them know about this thread. It would be good if someone could go over the driver here in details. At first glance, some stuff still needs proper support, and likely several fixes and many optimizations to be done.
Long story short, the zigate driver needs someone to take up the task of maintaining it; it pretty much stopped receiving updates 3 years ago... https://github.com/Koenkk/zigbee-herdsman/pulls?q=is%3Apr+zigate+in%3Atitle+

eswenson1, 2024-04-30:
I sent a message to my contact at ThirdReality and provided a link to this thread. Let’s see if it provokes any response.

eswenson1, 2024-05-01:
> I have a contact at ThirdReality who has been very responsible and helped me get things working. I hesitate to put his email address here, but it you want to contact me directly, I’ll give you his email.

I wanted to point out that you will see errors with HA/Zigbee2MQTT with the ThirdReality Zigbee USB dongle until you update to the latest firmware. The dongle came with older firmware, and Zigbee-herdsman reported errors parsing the status that came back from the dongle.

llhappier, 2024-05-06:
Hi.
The Zigate Protocol Introduction Link: https://zigate.fr/documentation/commandes-zigate/
I’m using Zigate with ThirdReality Zigbee USB dongle，also with works fine with HA OS/Zigbee2MQTT. 
It will report some errors but do not affect normal use.
I will take some time to review it. Thank you very much.

eswenson1, 2024-05-06:
I recently updated to the latest version of Zigbee2MQTT and now I can no longer successfully interact with my ThirdReality Zigbee USB Dongle.  The log shows messages like this:
```
[2024-05-05 15:44:02] info: 	zh:zigate:frame: {"msgCodeBytes":{"type":"Buffer","data":[132,1]},"msgLengthBytes":{"type":"Buffer","data":[0,12]},"checksumBytes":{"type":"Buffer","data":[204]},"msgPayloadBytes":{"type":"Buffer","data":[0,1,0,153,206,6,0,1,0,20,0]},"rssiBytes":{"type":"Buffer","data":[0]},"msgLengthOffset":-1}
[2024-05-05 15:44:02] error: 	zh:zigate:driver: Parsing error: Error: Message '8401' not found
[2024-05-05 15:44:04] info: 	zh:zigate:frame: {"msgCodeBytes":{"type":"Buffer","data":[128,2]},"msgLengthBytes":{"type":"Buffer","data":[0,23]},"checksumBytes":{"type":"Buffer","data":[106]},"msgPayloadBytes":{"type":"Buffer","data":[0,1,4,5,0,1,1,2,153,206,2,0,0,25,91,0,1,0,0,20,0,0]},"rssiBytes":{"type":"Buffer","data":[255]},"msgLengthOffset":-1}
[2024-05-05 15:44:04] info: 	zh:zigate:frame: {"msgCodeBytes":{"type":"Buffer","data":[132,1]},"msgLengthBytes":{"type":"Buffer","data":[0,12]},"checksumBytes":{"type":"Buffer","data":[204]},"msgPayloadBytes":{"type":"Buffer","data":[0,1,0,153,206,6,0,1,0,20,0]},"rssiBytes":{"type":"Buffer","data":[0]},"msgLengthOffset":-1}
[2024-05-05 15:44:04] error: 	zh:zigate:driver: Parsing error: Error: Message '8401' not found
```

Nerivec, 2024-05-06:
@eswenson1 Looks like frames that aren't implemented in the driver yet. That was only shown in `debug` mode before 1.37, so you wouldn't have seen these errors. I don't think that's the source of your problem. Can you upload a complete log file?

sebPomme, 2024-05-12:
Hi, i still use my [zigate+](https://zigate.fr/produit/zigate-usb/) it still works. I got some messages in the logs:
```
engine-1  | [2024-05-12 15:05:34] error:         zh:zigate:driver: Parsing error: Error: Message '8008' not found
```
 those messages brings me here
I use `koenkk/zigbee2mqtt:1.37.1`

mipsou, 2024-05-14:
Same error with same version

ttyUSB0
/dev/serial/by-id/usb-FTDI_ZiGate_ZIGATE+-if00-port0
Sous-système:
tty
Chemin d’accès au périphérique:
/dev/ttyUSB0
ID:
/dev/serial/by-id/usb-FTDI_ZiGate_ZIGATE+-if00-port0
Attributs:
DEVLINKS: >-
  /dev/serial/by-id/usb-FTDI_ZiGate_ZIGATE+-if00-port0
  /dev/serial/by-path/platform-fd8c0000.usb-usb-0:1:1.0-port0
  /dev/serial/by-path/platform-fd8c0000.usb-usbv1-0:1:1.0-port0
DEVNAME: /dev/ttyUSB0
DEVPATH: /devices/platform/fd8c0000.usb/usb5/5-1/5-1:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS: usb
ID_MODEL: ZiGate
ID_MODEL_ENC: ZiGate
ID_MODEL_ID: '6015'
ID_PATH: platform-fd8c0000.usb-usb-0:1:1.0
ID_PATH_TAG: platform-fd8c0000_usb-usb-0_1_1_0
ID_PATH_WITH_USB_REVISION: platform-fd8c0000.usb-usbv1-0:1:1.0
ID_REVISION: '1000'
ID_SERIAL: FTDI_ZiGate_ZIGATE+
ID_SERIAL_SHORT: ZIGATE+
ID_TYPE: generic
ID_USB_DRIVER: ftdi_sio
ID_USB_INTERFACES: ':ffffff:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: ZiGate
ID_USB_MODEL_ENC: ZiGate
ID_USB_MODEL_ID: '6015'
ID_USB_REVISION: '1000'
ID_USB_SERIAL: FTDI_ZiGate_ZIGATE+
ID_USB_SERIAL_SHORT: ZIGATE+
ID_USB_TYPE: generic
ID_USB_VENDOR: FTDI
ID_USB_VENDOR_ENC: FTDI
ID_USB_VENDOR_ID: '0403'
ID_VENDOR: FTDI
ID_VENDOR_ENC: FTDI
ID_VENDOR_ID: '0403'
MAJOR: '188'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'

Hedda, 2024-05-31:
> In https://github.com/Koenkk/zigbee-herdsman/pull/1033 some issues with the ZiGate adapter were discovered, some parts of the code look broken. Since I've not seen any issues reported about ZiGate, I'm wondering wether anyone is still using it and if it still works.

Perhaps merge this into -> https://github.com/Koenkk/zigbee-herdsman/issues/242

> Tagging some people who have worked on or used the ZiGate adapter

Maybe also ping @fairecasoimeme as he maintains original firmware and makes/sells ZiGate Coordinator (and LiXee) products?

https://github.com/fairecasoimeme

https://faire-ca-soi-meme.fr

https://zigate.fr

https://lixee.fr

> I have a contact at ThirdReality who has been very responsible and helped me get things working

FYI, I believe that @hwzolin and @weihuan1111 should also have contacts at ThirdReality (if they do not work for Third Reality?)?

devbis, 2024-05-31:
According to [docs](https://zigate.fr/documentation/commandes-zigate/), 8008 is just a hearbeat, so it can be safely ignored

incognito1234, 2024-06-08:
I still use it with 1 device, and 1 pending issue. I do not know if the root cause of the issue is zigate or the zlinky.

https://github.com/Koenkk/zigbee2mqtt/issues/22206

Hedda, 2024-06-17:
Off-topic but FYI, for reference; it sounds as if the zigpy project will soon also list their ZiGate radio adapter library (zigpy-zigate) as deprecated if no other willing developer steps up and volunteers to take over general maintenance and upkeep of its code of it:

https://github.com/zigpy/zigpy-zigate

https://github.com/zigpy/zigpy-zigate/issues/154

Hedda, 2024-10-21:
> I still have an original Zigate V1 in a drawer, so I could do some tests as well if needed.

ZiGate v1 is no longer maintained in upstream as far as I know, or is ZiGate v1 not decrepitated in upstream? The is a note there stating it is in ”maintenance mode” but the last ZiGate v1 firmware release is ZiGate v3.23 from October 17, 2022.

https://github.com/fairecasoimeme/ZiGate

So might be a good idea to only focus on ”ZiGate+” (ZiGate v2)?

https://github.com/fairecasoimeme/ZiGatev2

Though seeing that even that has not on February 21, 2023 it makes memwonder if the Zigbee Coordinator part for the ZiGate project have been abandoned?

All upstream focus in upstream is now focusing on LiXee-Box (which is more of a competitor of Zigbee2MQTT) and some Zigbee End Devices? 

https://github.com/fairecasoimeme?tab=repositories

closed 2024-12-29 by github-actions[bot]

pdecat, 2025-01-12:
FWIW, I've just recovered my old Zigate v1 (FW v3.2.3) from a drawer and installed Z2M with the following configuration:

```
version: '3.8'
services:
    mqtt:
        image: eclipse-mosquitto:2.0
        restart: unless-stopped
        volumes:
            - './mosquitto-data:/mosquitto'
        ports:
            - '1883:1883'
            - '9001:9001'
        command: 'mosquitto -c /mosquitto-no-auth.conf'

    zigbee2mqtt:
        container_name: zigbee2mqtt
        restart: unless-stopped
        image: koenkk/zigbee2mqtt
        volumes:
            - ./zigbee2mqtt-data:/app/data
            - /run/udev:/run/udev:ro
        ports:
            - 8080:8080
        environment:
            - TZ=Europe/Paris
        devices:
            - /dev/ttyUSB0:/dev/ttyUSB0
```

And it just works!

It managed to pair some new [NEO](https://www.zigbee2mqtt.io/supported-devices/#v=NEO) [NAS-STH02B2](https://www.zigbee2mqtt.io/devices/NAS-STH02B2.html#neo-nas-sth02b2) (TS0601 _TZE284_rqcuwlsa) device I'm testing without any issue.