Koenkk/zigbee-herdsman #1143: The readme docu link leads to 502

Matthias-Hermsdorf, 2024-08-08:
On the https://www.npmjs.com/package/zigbee-herdsman and on https://github.com/Koenkk/zigbee-herdsman [tsdocs.dev](https://tsdocs.dev/docs/zigbee-herdsman) is linked.

If I open it, it shows a CloudeFlare Bad Gateway Error.
It would be nice to see the documentation.

Nerivec, 2024-08-16:
Problems with the upstream service it seems (multiple issues already created there).

@Koenkk Maybe we could just use gh-pages branch here and deploy whenever a new version is published. Something along those lines (untested):

```yaml
name: Publish typedoc on Github Pages

on:
  release:
    types:
      - published

jobs:
  build:
    runs-on: ubuntu-latest
    permissions:
      contents: write
    steps:
      - uses: actions/checkout@v4
      - uses: actions/setup-node@v4
      - name: Generate docs
        run: |
          npm ci
          npm run docs
      - uses: actions/upload-pages-artifact@v3
        with:
          name: github-pages
          # typedoc "out" path
          path: ./docs
  deploy:
    needs: build
    runs-on: ubuntu-latest
    permissions:
      pages: write      # to deploy to Pages
      id-token: write   # to verify the deployment originates from an appropriate source
    environment:
      name: github-pages
      url: ${{ steps.deployment.outputs.page_url }}
    steps:
      - name: Deploy to GitHub Pages
        id: deployment
        uses: actions/deploy-pages@v4
```

and we change the `docs` script command to something like:
```json
"docs": "typedoc --gitRevision $(git describe --tag --abbrev=0) --tsconfig tsconfig.json --excludePrivate --excludeProtected --excludeExternals --entryPointStrategy expand ./src --sourceLinkTemplate 'https://github.com/Koenkk/zigbee-herdsman/blob/{gitRevision}/{path}#L{line}'"
```

Koenkk, 2024-08-17:
Fixed, docs are on https://koenkk.github.io/zigbee-herdsman/

Thanks @Nerivec

closed 2024-08-17 by Koenkk