Koenkk/zigbee-herdsman #1162: zh:ezsp:ezsp: Frame idConflictHandler parsing error: TypeError: Cannot set property id of #<EZSPFrameData> which has only a getter at new EZSPFrameData

saveriol, 2024-08-22:
Hello, I constantly see this issue while some of my devices become irresponsive (not sure if it's related):

`zh:ezsp:ezsp: Frame idConflictHandler parsing error: TypeError: Cannot set property id of #<EZSPFrameData> which has only a getter at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:177:23) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:151:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:145:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:430:35) at SerialDriver.emit (node:events:517:28) at SerialDriver.handleDATA (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:247:14) at SerialDriver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:172:26) at Parser.emit (node:events:517:28) at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:50:26)`

The only reference to idConflictHandler is [here](https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf):

>**Name**: idConflictHandler ID: 0x007C
>**Description**: A callback invoked by the EmberZNet stack when an id conflict is discovered, that is, two different nodes in the network were found to be using the same short id. The stack automatically removes the conflicting short id from its internal tables (address, binding, route, neighbor, and child tables). The application should discontinue any other use of the id.
This frame is a response to the callback command.
>**Response Parameters**:
>EmberNodeId id The short id for which a conflict was detected

saveriol, 2024-08-22:
Update: with ember driver the conflicts are correctly handled.

Koenkk, 2024-08-22:
> with ember driver the conflicts are correctly handled.

Then this can be closed, ezsp implementation is deprecated.

closed 2024-08-22 by Koenkk