Koenkk/zigbee-herdsman #1184: deconz driver/adapter (for ConBee/RaspBee Zigbee Coordinator) does not support joining/pairing via install code (or qr code)

Hedda, 2024-09-12:
deconz driver/adapter (for ConBee/RaspBee Zigbee Coordinator) [does not support joining/pairing via install code (or qr code)](https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20), but FYI; Dresden-Elektronik developer has now updated the deCONZ Serial Protocol documentation to v1.22 with info on install codes. See:

_Originally posted by @manup in https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20#issuecomment-2344518503_

_Sorry this took a while, I had reworked the document since the former .docx based one was messy to update. The document references are in a bit rough shape but that's for another version. The new version 1.22 has now been uploaded http://deconz.dresden-elektronik.de/raspbian/deCONZ-Serial-Protocol_en_1.22.pdf_

_There is a new small section **6.2.3 Zigbee Install Codes** which describes the basic usage._

_As reference how it works in deCONZ the following sources might be helpful as well:_

_Calculating the derived Link Key from Zigbee Install Code (I think ZHA and Z2M already have similar code):
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/crypto/mmohash.cpp_

_How the install code is processed by the REST-API and set to the firmware:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/0e2dc155e85893b439f354cd5d764e414ad0ed0c/rest_devices.cpp#L1090_

_Originally posted by @manup in https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20#issuecomment-2344518503_

Maginos2210, 2024-09-13:
@Hedda: Thank you for the information.
I'm totally new in the field of home automation, so forgive my beginner question:

Is there any guide on what I have to do now to get my Bosch Thermostat 2 working with a conbee II stick? 
How can I calculate the Link Key from the file mmohash.cpp? And what do I have to do then?

Thank you for your help.

Maginos

Hedda, 2024-09-13:
> Is there any guide on what I have to do now to get my Bosch Thermostat 2 working with a conbee II stick?

As long as you using any ConBee adapter then you are out of luck until install codes function is added the deconz driver here. See:

* https://www.zigbee2mqtt.io/guide/faq/#general-limitations-that-apply-to-all-zigbee-implementations

"_**Support for commissioning Zigbee 3.0 devices via "Install Code" or "QR Code" has so far only been implemented for 'zstack' (Texas Instruments ZNP) and 'ember' (Silicon Labs EmberZNet) adapter type radios in Zigbee2MQTT. Other radio adapter types are either missing support in their respective adapter/driver for [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman) or more likely missing in the manufacturer's firmware commands/APIs and documentation.**_"

It means that unless you are a developer that program and can add that feature to the [deconz driver/adapter code in zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman/tree/master/src/adapter/deconz) then there is nothing you can do as long as you are using a ConBee (or RaspBee) adapter as your Zigbee Coordinator.

So if you are an end-users looking for a work around that today without it being supported in the deconz driver/adapter you have to migrate to either a 'zstack' (Texas Instruments ZNP) or 'ember' (Silicon Labs EmberZNet) adapter, ...but there is a catch with migrations to/from deconz adapters as well because the [deconz driver/adapter in zigbee-herdsman (and thus Zigbee2MQTT) does yet not support backup and restore for Zigbee network from or to ConBee/RaspBee](https://github.com/Koenkk/zigbee-herdsman/issues/665), and therefore you currently would need to [use an unofficial third-party external tools (like example zigpy-cli) to backup and restore your Zigbee network from ConBee to a Texas Instruments ZNP or Silicon Labs EmberZNet adapter](https://github.com/Koenkk/zigbee2mqtt/discussions/16478) (which is doable but in turn [will require you to also re-pair all your devices regardless](https://www.zigbee2mqtt.io/guide/faq/#what-does-and-does-not-require-repairing-of-all-devices)). Please respect that this is not the place to to discuss such workarounds.

> How can I calculate the Link Key from the file mmohash.cpp? And what do I have to do then?

While related I believe that question too is also off-topic here in regards to specifically calculate Link Key for Bosch Thermostat 2 if you are an end-users that is now no longer using the deconz driver/adapter with ConBee/RaspBee as have instead migrated to Texas Instruments ZNP or Silicon Labs EmberZNet adapter. Regardless we do not want to spam this zigbee-herdsman feature request issue with discussions about device specific questions.

Hence, if you are and end-user and go the way of migrating to Texas Instruments ZNP or Silicon Labs EmberZNet adapter and then still need help calculating Link Keys for your Bosch Thermostat 2 devices then please post in the Zigbee2MQTT's discussions instead -> https://github.com/Koenkk/zigbee2mqtt/discussions (or in zigbee-herdsman-converters if related to changing or extending the zigbee-herdsman-converter for that device -> https://github.com/Koenkk/zigbee-herdsman-converters/issues ).

Maginos2210, 2024-09-13:
Thank you @Hedda for your explanations. I ordered a Sonoff zigbee stick with a Texas Instruments chip. With this, it should work. Migration is not needed, since the Bosch thermostat is my first device.

Hedda, 2024-09-13:
@ChrisHae @manup @Haerteleric any chance one of you guys would be interested and willing to spare some of your time to take a closer look at this requested feature/function of adding support for "install codes" provisioning in deconz driver (for ConBee/RaspBee Zigbee Coordinator adapters) inside the [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman) framework so that [Zigbee2MQTT](https://www.zigbee2mqtt.io) users with  ConBee/RaspBee adapters can provision Zigbee devices that require install code or QR code for commissioning?

markusmauch, 2024-10-12:
It looks to me as if Dresden Electronik already has implemented [Zigbee Install Codes](https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20#issuecomment-2344518503). 
Is it realistic to hope to see this implemented for a Conbee II stick and Zigbee2MQTT in the near future?

Hedda, 2024-11-08:
> It looks to me as if Dresden Electronik already has implemented [Zigbee Install Codes](https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20#issuecomment-2344518503).

FYI, it not clear but also sounds as if zigpy/zha developers have implemeneted it too as closed issue -> https://github.com/zigpy/zigpy-deconz/issues/214

> Is it realistic to hope to see this implemented for a Conbee II stick and Zigbee2MQTT in the near future?

That depends if Dresden-Elektronik's own developers submit PR for it to zigbee-herdsman or some other volunteering steps-up.

Suggest that you maybe post to ask in their phoscon.de forum at https://forum.phoscon.de/c/general/5 as not been replying here.

Koenkk, 2024-11-24:
@Nerivec added support for this in https://github.com/Koenkk/zigbee-herdsman/pull/1243, neither @Nerivec or me have a device to test this with so would be great if someone can test this. This change will be available in the dev branch after the 1 December release.

SoftwareSchlosser, 2024-12-01:
Just tried to add a Bosch thermostat (using ConBee II) and this is what I get:

```
z2m: Error: Add install code is not supported
    at DeconzAdapter.addInstallCode (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:312:37)
    at Controller.addInstallCode (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:274:28)
    at Zigbee.addInstallCode (/app/lib/zigbee.ts:397:29)
    at Bridge.installCodeAdd (/app/lib/extension/bridge.ts:325:27)
    at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:200:63)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:144:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:150:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:170:27)

```

Nerivec, 2024-12-01:
As Koenkk mentioned, support will be in dev branch. It won't be in release branch until next release.

SoftwareSchlosser, 2024-12-01:
Sorry, forgot to mention that in fact I'm using the `koenkk/zigbee2mqtt:latest-dev` docker image.

Nerivec, 2024-12-01:
2.0.0 is not yet available in dev branch, have to wait until the release process is over, lots of moving pieces on this one.

SoftwareSchlosser, 2024-12-01:
OK then I misunderstood [this post](1184#issuecomment-2496220806):
> This change will be available in the dev branch after the 1 December release.

Didn't know there is a 2.0.0 dev-branch besides the 1.x dev-branch.
Thank you for clarifying that, I'll just wait then ;)

Koenkk, 2024-12-01:
It's available in the dev branch now.

Nerivec, 2024-12-02:
Can you provide the rest of the logs after that, to see what happens exactly?

SoftwareSchlosser, 2024-12-02:
I didn't see any related log messages this morning but after re-trying it right now I noticed a timeout error after 10s:

```
[2024-12-02 18:56:47] debug: 	zh:deconz:driver: Timeout for request - CMD: 0xb seqNr: 1
[2024-12-02 18:56:47] error: 	z2m: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'TIMEOUT'
[2024-12-02 18:56:47] debug: 	z2m: Error: TIMEOUT
    at Driver.processBusyQueue (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:508:28)
    at Timeout._onTimeout (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:160:46)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
[2024-12-02 18:56:47] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"TIMEOUT","status":"error","transaction":"md2q6-1"}'
```

Hard to tell if there's more because there is a lot of traffic on my zigbee network and it's hard to separate what message belongs to which device/event.

Nerivec, 2024-12-02:
Are you seeing this timeout every time? If it didn't work, I'd expect an error response, but not a timeout.

Worth mentioning that deconz's API considers adding install codes as experimental...
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/0e2dc155e85893b439f354cd5d764e414ad0ed0c/rest_devices.cpp#L1088
Not sure how that fares for support on the firmware side...

Hedda, 2024-12-03:
> Worth mentioning that deconz's API considers adding install codes as experimental...
> https://github.com/dresden-elektronik/deconz-rest-plugin/blob/0e2dc155e85893b439f354cd5d764e414ad0ed0c/rest_devices.cpp#L1088
> Not sure how that fares for support on the firmware side...

Tip is that @manup have replied to direct questions posted in their repository here -> https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20

gramels, 2024-12-08:
I do get the same errors as @SoftwareSchlosser .
Adding Install Code, then time out.

Nerivec, 2024-12-08:
@gramels Are you using a Conbee II also? Can you provide a full `debug` log from the moment you add the install code, to a few seconds after the timeout? We need to see if there is a special response coming that isn't currently handled by the driver.

gramels, 2024-12-10:
`2024-12-10 11:19:00zh:controller: Adding install code for xxx.
Info 2024-12-10 11:19:08z2m: Successfully added new install code
Info 2024-12-10 11:19:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{"value":"xxx"},"status":"ok","transaction":"o3hyh-1"}'
Info 2024-12-10 11:19:08z2m: Zigbee: allowing new devices to join.
Info 2024-12-10 11:19:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"o3hyh-2"}'`

I tried now from remote (without the Bosch TRV) in pairing mode and did *not* get a time out. So as it appears the time out is a strange response. I use a conbee III (3).

Nerivec, 2024-12-10:
We're going to need a full `debug` log of adding the code + pairing (or timeout) and likely a sniff of the zigbee network to figure out what's going on with the adapter. If someone can provide that, let me know (we can exchange the data in private if preferred).

gramels, 2024-12-10:
I enabled debug on z2m, thats what I got. Where would I enable/ extract this (on hass os)?

Nerivec, 2024-12-10:
It does not appear you have enabled `debug` level from your last logs, this is `info` level (missing lines like `send write parameter request`).

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html for details on sniffing (requires a spare adapter to sniff with, easiest currently being [EmberZNet-based ones](https://github.com/Nerivec/ember-zli/wiki/Sniff)). It's done separately from Z2M/HA, any computer will do.

raoulbhatia, 2024-12-31:
I am also trying to add Bosch devices to my z2m installation
but receive `z2m: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'TIMEOUT'`

Please let me know if I can provide more details.

### Setup

I am using a Home Assistant Green with
* HAOS, **version 14.1**
* HA, **version 2024.12.5**
* **Mosquitto broker** add-on, **version: 6.4.1**
* **Zigbee2MQTT Edge** add-on, **version: edge**

### Zigbee2MQTT
* Zigbee2MQTT version: 1.42.0-dev commit: [62e18a6](https://github.com/Koenkk/zigbee2mqtt/commit/62e18a6)
* Coordinator type: ConBee3
* Coordinator revision: 0x26530900 (latest version from 11-Jul-2024 from [deconz-firmware](https://deconz.dresden-elektronik.de/deconz-firmware/))
* zigbee-herdsman-converters version: 21.9.2
* zigbee-herdsman version: 3.1.1

### Logs
1. I've enabled `zh:debug` mode
1. Please note that I am also running a few SNZB-02 devices in this setup

```
[2024-12-31 10:00:56] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 101010
[2024-12-31 10:00:56] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-12-31 10:00:56] debug: 	zh:deconz:driver: query aps data indication
[2024-12-31 10:00:56] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 95
[2024-12-31 10:00:56] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 95 srcAddr: 0xaecc destAddr: 0x0 profile id: 0x104 cluster id: 0x402 lqi: 200
[2024-12-31 10:00:56] debug: 	zh:deconz:frameparser: response payload: 185f0a000029e207
[2024-12-31 10:00:56] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-12-31 10:00:56] debug: 	zh:controller: Received payload: clusterID=1026, address=44748, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=200, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":95,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":41,"attrData":2018}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-12-31 10:00:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SNZB-02 1D1F', payload '{"battery":95.5,"humidity":52.29,"linkquality":200,"temperature":20.18,"voltage":2900}'
[2024-12-31 10:00:59] debug: 	zh:deconz:driver: send read device state from queue. seqNr: 96
[2024-12-31 10:00:59] debug: 	zh:deconz:frameparser: device state: 100010
[2024-12-31 10:00:59] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-12-31 10:01:04] info: 	zh:controller: Install code was adjusted for reason 'missing CRC'.
[2024-12-31 10:01:04] info: 	zh:controller: Adding install code for 0x000D6F0019933538.
[2024-12-31 10:01:04] debug: 	zh:deconz:driver: send write parameter request from queue. seqNr: 97 paramId: 25 param: 56,53,147,25,0,111,13,0,104,36,80,155,167,113,54,189,19,236,247,91,95,89,131,141
[2024-12-31 10:01:14] debug: 	zh:deconz:driver: Timeout for request - CMD: 0xb seqNr: 97
[2024-12-31 10:01:14] error: 	z2m: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'TIMEOUT'
[2024-12-31 10:01:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"TIMEOUT","status":"error","transaction":"izmwx-4"}'
[2024-12-31 10:01:14] debug: 	zh:deconz:driver: send read device state from queue. seqNr: 98
[2024-12-31 10:01:14] debug: 	zh:deconz:frameparser: device state: 100010
[2024-12-31 10:01:14] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-12-31 10:01:19] debug: 	zh:deconz:driver: send read device state from queue. seqNr: 99
[2024-12-31 10:01:19] debug: 	zh:deconz:frameparser: device state: 100010
[2024-12-31 10:01:19] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-12-31 10:01:29] debug: 	zh:deconz:driver: send read device state from queue. seqNr: 100
[2024-12-31 10:01:29] debug: 	zh:deconz:frameparser: device state: 100010
[2024-12-31 10:01:29] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-12-31 10:01:38] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 101010
[2024-12-31 10:01:38] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-12-31 10:01:38] debug: 	zh:deconz:driver: query aps data indication
[2024-12-31 10:01:38] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 101
[2024-12-31 10:01:38] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 101 srcAddr: 0x6365 destAddr: 0x0 profile id: 0x104 cluster id: 0x405 lqi: 194
[2024-12-31 10:01:38] debug: 	zh:deconz:frameparser: response payload: 18130a0000218c16
[2024-12-31 10:01:38] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-12-31 10:01:38] debug: 	zh:controller: Received payload: clusterID=1029, address=25445, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=194, frame={"header":{"frameControl":{"frame [... cut ...] ,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":245,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":33,"attrData":5670}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-12-31 10:01:49] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SNZB-02 1147', payload '{"battery":91,"humidity":56.7,"linkquality":183,"temperature":20.02,"voltage":2900}'
```

raoulbhatia, 2025-01-05:
@Nerivec @Koenkk please let me know what I can do to help get the install code to work in z2m 😊

Nerivec, 2025-01-06:
A little strange that it would take >10s and then sometimes succeeds in less than 5sec (pretty slow in any case though).
Could be linked to load status of the NCP I suppose, if it does some crypto op with the install code, that definitely could drag its perf way down until it's done...

Can you reproduce the successful add reliably?
Can you also confirm if the device pairs successfully after a successful add?

raoulbhatia, 2025-01-06:
This morning, after >10h of not touching anything,
I tried to add the install code again and this succeeded (logs below, though without debug)

However, I did NOT manage to pair the Twinguard.

I will enable debug logs and try again in a few hours.
Could you please let me know if `zh` debug logs are sufficient, or if I better switch everything to `debug`?

```
[2025-01-06 08:08:15] info: 	zh:controller: Install code was adjusted for reason 'missing CRC'.
[2025-01-06 08:08:15] info: 	zh:controller: Adding install code for 0x000D6F0019933538.
[2025-01-06 08:08:19] info: 	z2m: Successfully added new install code
[2025-01-06 08:08:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{"value":"RB01SG0D836591B3CC0010000000000000000000000D6F0019933538DLK07C0A885D6107EE86D5BC3248658B27E"},"status":"ok","transaction":"681jv-1"}'
[2025-01-06 08:08:19] info: 	z2m: Zigbee: allowing new devices to join.
[2025-01-06 08:08:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"681jv-2"}'
```

raoulbhatia, 2025-01-06:
I tried to add the install code again after a few hours.

The first attempt did not succeed.  What I find interesting is the following logs hinting at

https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/deconz/driver/driver.ts#L508

I did not dig into the timeoutcounter logic,
but perhaps there is a relation to the problem at hand?

```
[2025-01-06 11:27:00] debug: 	z2m: Error: TIMEOUT
    at Driver.processBusyQueue (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:508:28)
    at Timeout._onTimeout (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:160:46)
    at listOnTimeout (node:internal/timers:594:17)
    at processTimers (node:internal/timers:529:7)
```

After changing the log level again to ensure EVERYTHING Is logged at `debug` level,
I was able to successfully add the install code.

However, the device did not manage to join.
You can find the logs at https://nem3d.net/z2m_debug_20240106a.txt

FYI I am running my Zigbee network on channel 25.
But from what I read this should also be supported by Bosch's own controller.

raoulbhatia, 2025-01-31:
What else can I do to help with this issue?

Koenkk, 2025-02-10:
should be fixed by https://github.com/Koenkk/zigbee-herdsman/pull/1317, could you try again on the latest dev branch?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

calumljohnston, 2025-02-28:
Just tried again and still getting timeout error.

Zigbee2MQTT version: 2.1.1-dev commit: [102857a7fec60e3f65887d92b6fe7a2d81add110](https://github.com/Koenkk/zigbee2mqtt/commit/102857a7fec60e3f65887d92b6fe7a2d81add110)
Coordinator type: ConBee2/RaspBee2
Coordinator revision: 0x26780700
Coordinator IEEE Address: 0x00212effff060853
Frontend version: 0.9.4
zigbee-herdsman-converters version: 23.1.1
zigbee-herdsman version: 3.2.7

Logs:
```
debug 2025-02-28 15:07:31zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2025-02-28 15:07:34zh:deconz:driver: send read device state from queue. seqNr: 247
debug 2025-02-28 15:07:34zh:deconz:frameparser: device state: 10100010
debug 2025-02-28 15:07:34zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2025-02-28 15:07:34zh:deconz:frameparser: device state: 10100010
debug 2025-02-28 15:07:34zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2025-02-28 15:07:40z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/install_code/add' with data '{"transaction":"bst4w-13","value":"RB01SG0D83101826480080000000000000000000C498860000D4C4A7DLK6B20E175AAD3F960FB397B5509BBAF86A3AF"}'
info 2025-02-28 15:07:40zh:controller: Adding install code for 0xC498860000D4C4A7.
debug 2025-02-28 15:07:40zh:deconz:driver: send write parameter request from queue. seqNr: 152 paramId: 25 param: 167,196,212,0,0,134,152,196,110,93,33,176,28,116,217,104,181,178,0,250,107,36,217,11
debug 2025-02-28 15:07:50zh:deconz:driver: Timeout for request - CMD: 0xb seqNr: 152
debug 2025-02-28 15:07:50zh:deconz:driver: too many timeouts - restart serial connecion
debug 2025-02-28 15:07:50zh:deconz:driver: Opening with /dev/ttyACM0
error 2025-02-28 15:07:50z2m: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'TIMEOUT'
debug 2025-02-28 15:07:50z2m: Error: TIMEOUT at Driver.processBusyQueue (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:508:28) at Timeout._onTimeout (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:160:46) at listOnTimeout (node:internal/timers:594:17) at processTimers (node:internal/timers:529:7)
info 2025-02-28 15:07:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{},"error":"TIMEOUT","status":"error","transaction":"bst4w-13"}'
debug 2025-02-28 15:07:50zh:deconz:driver: Serialport opened
debug 2025-02-28 15:07:50zh:deconz:driver: send read device state from queue. seqNr: 7
debug 2025-02-28 15:07:50zh:deconz:frameparser: device state: 10100010
debug 2025-02-28 15:07:50zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2025-02-28 15:07:50zh:deconz:frameparser: device state: 10100010
debug 2025-02-28 15:07:50zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2025-02-28 15:07:50zh:deconz:frameparser: device state: 10100010
debug 2025-02-28 15:07:50zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2025-02-28 15:07:50zh:deconz:driver: send read device state from queue. seqNr: 153
debug 2025-02-28 15:07:50zh:deconz:frameparser: device state: 10100010
```

flue17, 2025-06-30:
Tried it with the latest dev branch and successfully paired a Bosch Thermostat 2 via install code.

```
[2025-06-30 19:09:49] zh:controller: Succesfully interviewed '0x18fc2600000a9c1b'

[2025-06-30 19:09:49] z2m: Successfully interviewed '0x18fc2600000a9c1b', device has successfully been paired

[2025-06-30 19:09:49] z2m: Device '0x18fc2600000a9c1b' is supported, identified as: Bosch Radiator thermostat II (BTH-RA)

[2025-06-30 19:09:49] z2m: Configuring '0x18fc2600000a9c1b'

Koenkk, 2025-06-30:
Dev branch indeed contains a completely refactored Deconz implementation, please try with that! (will be included in tomorrows new release)

closed 2025-07-02 by Koenkk