Koenkk/zigbee-herdsman #1220: Issue with network scan

asgothian, 2024-10-20:
When swapping the zigbee-herdsman from 0.56.2 to 2.1.1, the functionality used for network scan dies.

Code:
```
    async getChannelsEnergy() {
        const payload = {
            dstaddr: 0x0,
            dstaddrmode: 0x02,
            channelmask: 0x07FFF800,
            scanduration: 0x5,
            scancount: 1,
            nwkmanageraddr: 0x0000
        };
        const energyScan = this.herdsman.adapter.znp.waitFor(
            2, // unpi_1.Constants.Type.AREQ,
            5, // Subsystem.ZDO,
            'mgmtNwkUpdateNotify'
        );
        await this.herdsman.adapter.znp.request(
            0x5, // Subsystem.ZDO
            'mgmtNwkUpdateReq',
            payload,
            energyScan.ID
        );
        const result = await energyScan.start().promise;
        return result.payload;
    }

```

Error message when this code is run with the herdsman 2.1.1 to 2.1.4:
```
2024-10-20 09:58:24.352 - error: host.raspberrypi Caught by controller[2]: Error: Command request 'mgmtNwkUpdateReq' from subsystem '5' not found
```

Tested with various firmware versions and drivers.

A.

Koenkk, 2024-10-20:
This has been refactored to ZDO, use send a ZDO request instead: [example](https://github.com/Koenkk/zigbee-herdsman/blob/425e58ec8e1d1481bb8bb8dead0a0e71c3aef0c3/src/controller/controller.ts#L517)

asgothian, 2024-10-20:
Thanks for the hint. Will do. Closing this for now.

A.

closed 2024-10-20 by asgothian

Koenkk, 2024-10-21:
Look at the intelli sense of the [sendZdo](https://github.com/Koenkk/zigbee-herdsman/blob/425e58ec8e1d1481bb8bb8dead0a0e71c3aef0c3/src/controller/controller.ts#L517) here, the arguments are almost the same.

asgothian, 2024-10-21:
thanks for the heads up - will do.

A.

asgothian, 2024-10-24:
I can add a bit more info to it:

When changing the two const from -1 to 0, I was able to scan channels, but I never managed to do a full scan

this code 
```
let zdoPayload1 = ZigbeeHerdsmanZDO.Buffalo.buildRequest(true, clusterId, [11,12,13,14,15,16,17,18], 0x5, undefined, 0, undefined);
const res1 = await this.herdsman.adapter.sendZdo(0x0, 0x0, clusterId , zdoPayload1, false);

```
returns data

this code 
```
let zdoPayload1 = ZigbeeHerdsmanZDO.Buffalo.buildRequest(true, clusterId, [16,17,18,19,20,21], 0x5, undefined, 0, undefined);
const res1 = await this.herdsman.adapter.sendZdo(0x0, 0x0, clusterId , zdoPayload1, false);

```
returns goes into timeout after 10 seconds

as does this code
```
let zdoPayload1 = ZigbeeHerdsmanZDO.Buffalo.buildRequest(true, clusterId, [11,12,13,14,15,16,17,18,19], 0x5, undefined, 0, undefined);
const res1 = await this.herdsman.adapter.sendZdo(0x0, 0x0, clusterId , zdoPayload1, false);

```

I did try a number of variants with channel masks and so on - none of them worked. Although on some calls.i got a timeout after 6 seconds.

I am not sure how to solve this.

A.

Koenkk, 2024-10-24:
Could you sniff and see if the correct request is being send out?

asgothian, 2024-10-24:
I’m afraid I can’t - I am out traveling for the next few weeks with only limited hardware.

@arteck - can you ?

arteck, 2024-10-26:
I am looking for the cc2531

arteck, 2024-10-28:
so... new information from here

i see communication on chanel 11

[image]

but not on chanel 12 
i hope the source is correct
```
let payload = ZDO.Buffalo.buildRequest(false, clusterId, [12], 0x05, undefined, 0, undefined);
            this.warn(`Payload is [${JSON.stringify(payload)}]`);
            result = await this.herdsman.adapter.sendZdo(0x0, 0x0, clusterId , payload, false);
```

[image]

Koenkk, 2024-11-02:
Could you compare the diff in serial data send to the coordinator vs older versions?

asgothian, 2024-11-02:
I can check if I can catch the command sent to the coordinator. Unfortunately, I cannot sniff serial data, so I can't check what is actually sent via the serial link.

A.

asgothian, 2024-11-03:
Fixed. A closer analysis of the function building the payload has led me to the following code, which does generate a working scan:

```
        const clusterId = ZDO.ClusterId.NWK_UPDATE_REQUEST;
        const result = {};
        try
        {
            const payload = ZDO.Buffalo.buildRequest(false, clusterId, [11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26], 0x05, 1, 0, undefined);
            const scanresult = await this.herdsman.adapter.sendZdo(0x0, 0x0, clusterId , payload, false);
            result.energyvalues = scanresult[1].entryList;
        }
        catch (error) {
            this.sendError(error);
            this.error(`Failed to scan channels ${error.stack}`);
            result.error = error;

        }
        return result;

    }

```