Koenkk/zigbee-herdsman #1288: Random crashes with zcl message timeouts

asgothian, 2025-01-08:
Hi,

since changing to the zigbee-herdsman 2.x, I am seeing more and more cases of sudden adapter crashes with the following characteristics:
- the entire call-stack for the error is within the zigbee-herdsman, there is no evidence of a call from the iobroker.zigbee Adapter
- the crashes appear 'out of the blue' and are usually connected to one specific device on the zigbee Network.
- removing the device from the network removes the crashes
- adding the (identical) device again **can** fix the issue temporarily (not enough data to be certain when it does and when it does not)
- a larger range of devices are affected. (see [this issue](https://github.com/ioBroker/ioBroker.zigbee/issues/2353) on the iobroker.zigbee adapter for one example of a Tuya device)

I have so far refrained from opening an issue as I was not directly affected and as such had no 'test device'. Now it seems I do. Below is the log section which contains the error. In this case, the affected device is an Ikea transformer (ICPSHC24-10EU-IL-1)

I have checked the zigbee-adapter code to see if any call to a herdsman function is not enclosed in try/catch. I have not found a single one.

What I could see in the logs that the device in question seems to send a large number (comparatively) of 'device announce' messages, despite being constantly connected to power. This may indicate an issue with the device itself - but even if that is the case, a malfunctioning device should not crash the zigbee system.

```
2025-01-07 21:22:03.890 - info: zigbee.0 (4987) Device '0x000b57fffeb21684' announced itself
2025-01-07 21:22:07.182 - info: zigbee.0 (4987) Device '0x000b57fffeb21684' announced itself
2025-01-07 21:22:17.716 - error: zigbee.0 (4987) Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
2025-01-07 21:22:17.717 - error: zigbee.0 (4987) unhandled promise rejection: ZCL command 0x000b57fffeb21684/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 15611 - 1 - 48 - 6 - 7 after 10000ms)
2025-01-07 21:22:17.717 - error: zigbee.0 (4987) Error: ZCL command 0x000b57fffeb21684/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 15611 - 1 - 48 - 6 - 7 after 10000ms)
at Object.start (/opt/iobroker/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:640:47)
at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:672:28)
at /opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20
at Queue.execute (/opt/iobroker/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
at ZStackAdapter.sendZclFrameToEndpoint (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)
at Request.send (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
at Endpoint.sendRequest (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
at Endpoint.zclCommand (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
at Endpoint.configureReporting (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9)
2025-01-07 21:22:17.717 - error: zigbee.0 (4987) ZCL command 0x000b57fffeb21684/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 15611 - 1 - 48 - 6 - 7 after 10000ms)
2025-01-07 21:22:17.729 - info: zigbee.0 (4987) cleaned everything up...
2025-01-07 21:22:17.732 - info: zigbee.0 (4987) Zigbee: disabling joining new devices.
2025-01-07 21:22:18.230 - info: zigbee.0 (4987) terminating
2025-01-07 21:22:18.232 - warn: zigbee.0 (4987) Terminated (UNCAUGHT_EXCEPTION): Without reason
2025-01-07 21:22:18.863 - error: host.raspberrypi Caught by controller[1]: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
2025-01-07 21:22:18.865 - error: host.raspberrypi Caught by controller[2]: Error: ZCL command 0x000b57fffeb21684/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 15611 - 1 - 48 - 6 - 7 after 10000ms)
2025-01-07 21:22:18.866 - error: host.raspberrypi Caught by controller[2]: at Object.start (/opt/iobroker/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
2025-01-07 21:22:18.866 - error: host.raspberrypi Caught by controller[2]: at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:640:47)
2025-01-07 21:22:18.866 - error: host.raspberrypi Caught by controller[2]: at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:672:28)
2025-01-07 21:22:18.866 - error: host.raspberrypi Caught by controller[2]: at /opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20
2025-01-07 21:22:18.867 - error: host.raspberrypi Caught by controller[2]: at Queue.execute (/opt/iobroker/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
2025-01-07 21:22:18.867 - error: host.raspberrypi Caught by controller[2]: at ZStackAdapter.sendZclFrameToEndpoint (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)
2025-01-07 21:22:18.867 - error: host.raspberrypi Caught by controller[2]: at Request.send (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
2025-01-07 21:22:18.867 - error: host.raspberrypi Caught by controller[2]: at Endpoint.sendRequest (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
2025-01-07 21:22:18.867 - error: host.raspberrypi Caught by controller[2]: at Endpoint.zclCommand (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
2025-01-07 21:22:18.868 - error: host.raspberrypi Caught by controller[2]: at Endpoint.configureReporting (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9)
2025-01-07 21:22:18.868 - error: host.raspberrypi instance system.adapter.zigbee.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2025-01-07 21:22:18.869 - info: host.raspberrypi Restart adapter system.adapter.zigbee.0 because enabled
2025-01-07 21:22:49.453 - info: host.raspberrypi instance system.adapter.zigbee.0 in version "1.10.14" started with pid 6235
2

```

Device signature on startup:
`2025-01-07 21:22:55.955 - info: zigbee.0 (6235) 0x000b57fffeb21684 (addr 15611): ICPSHC24-10EU-IL-1/ICPSHC24-10EU-IL-2 - IKEA TRADFRI LED driver, 10 w (Router)`

asgothian, 2025-01-08:
Attached a log from today - the offending device is an Ikea lightbulb. At the time of the crashes, the bulb was flickering, turning off for a split second and then turning back on, indicating an error with the hardware - still the software should be able to handle this without crashing out.

```
2025-01-08 16:33:06.359 - info: zigbee.0 (7796) Device '0x000b57fffe8f487d' announced itself
2025-01-08 16:33:06.449 - error: zigbee.0 (7796) Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
2025-01-08 16:33:06.449 - error: zigbee.0 (7796) unhandled promise rejection: ZCL command 0x000b57fffe8f487d/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))
2025-01-08 16:33:06.450 - error: zigbee.0 (7796) Error: ZCL command 0x000b57fffe8f487d/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))
at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:524:23)
at /opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20
at Queue.execute (/opt/iobroker/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
at ZStackAdapter.sendZclFrameToEndpoint (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)
at Request.send (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
at Endpoint.sendRequest (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
at Endpoint.zclCommand (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
at Endpoint.configureReporting (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9)
at bulbOnEvent (/opt/iobroker/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:54:17)
at Object.onEvent (/opt/iobroker/node_modules/zigbee-herdsman-converters/src/index.ts:214:21)
2025-01-08 16:33:06.450 - error: zigbee.0 (7796) Exception-Code: 25: ZCL command 0x000b57fffe8f487d/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))
2025-01-08 16:33:06.471 - info: zigbee.0 (7796) cleaned everything up...
2025-01-08 16:33:06.474 - info: zigbee.0 (7796) Zigbee: disabling joining new devices.
2025-01-08 16:33:06.972 - info: zigbee.0 (7796) terminating
2025-01-08 16:33:06.973 - warn: zigbee.0 (7796) Terminated (UNCAUGHT_EXCEPTION): Without reason
2025-01-08 16:33:07.550 - error: host.raspberrypi Caught by controller[1]: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
2025-01-08 16:33:07.551 - error: host.raspberrypi Caught by controller[1]: Error: ZCL command 0x000b57fffe8f487d/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))
2025-01-08 16:33:07.551 - error: host.raspberrypi Caught by controller[1]: at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:524:23)
2025-01-08 16:33:07.556 - error: host.raspberrypi Caught by controller[1]: at /opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20
2025-01-08 16:33:07.556 - error: host.raspberrypi Caught by controller[1]: at Queue.execute (/opt/iobroker/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
2025-01-08 16:33:07.556 - error: host.raspberrypi Caught by controller[1]: at ZStackAdapter.sendZclFrameToEndpoint (/opt/iobroker/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)
2025-01-08 16:33:07.556 - error: host.raspberrypi Caught by controller[1]: at Request.send (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
2025-01-08 16:33:07.557 - error: host.raspberrypi Caught by controller[1]: at Endpoint.sendRequest (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
2025-01-08 16:33:07.557 - error: host.raspberrypi Caught by controller[1]: at Endpoint.zclCommand (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
[... cut ...]
2025-01-08 16:34:50.390 - error: host.raspberrypi Caught by controller[5]: at Object.onEvent (/opt/iobroker/node_modules/zigbee-herdsman-converters/src/index.ts:214:21)
2025-01-08 16:34:50.390 - error: host.raspberrypi instance system.adapter.zigbee.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2025-01-08 16:34:50.390 - info: host.raspberrypi Restart adapter system.adapter.zigbee.0 because enabled
2025-01-08 16:34:50.391 - warn: host.raspberrypi Do not restart adapter system.adapter.zigbee.0 because restart loop detected
```

Koenkk, 2025-01-08:
I've not seen such reports with z2m yet. Is the `onEvent` properly awaited like z2m does here ?

https://github.com/Koenkk/zigbee2mqtt/blob/060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a/lib/extension/onEvent.ts#L50

asgothian, 2025-01-09:
I am afraid I do not follow you. The 'onEvent' is not called directly from the zigbee adapter code. Instead, the adapter subscribes to a number of events :

```
            this.herdsman.on('adapterDisconnected', this.handleDisconnected.bind(this));
            this.herdsman.on('deviceAnnounce', this.handleDeviceAnnounce.bind(this));
            this.herdsman.on('deviceInterview', this.handleDeviceInterview.bind(this));
            this.herdsman.on('deviceJoined', this.handleDeviceJoined.bind(this));
            this.herdsman.on('deviceLeave', this.handleDeviceLeave.bind(this));
            this.herdsman.on('message', this.handleMessage.bind(this));
            this.herdsman.on('permitJoinChanged', this.handlePermitJoinChanged.bind(this));

```

The various functions go into different additional functions, but those are not called in case of an error - I enabled debug to see if any call was made, but there was not. I also checked all those functions, but any relevant call to the zigbee-herdsman is at this point enclosed in try/catch. I say relevant, because there are simple calls to e.g. getPermitJoin which are not enclosed in its own try/catch, but I do not see how these would be connected to the error message.

I am currently working off the assumption that onEvent calls made from converters on their own timing may be the root cause of the error, but that is deep within code I cannot actively debug.

A.
p.s. I had to move the bulb from my live zigbee network to my development network - now the bulb is quiet again. I will update here once I have more information.

Koenkk, 2025-01-10:
> I am currently working off the assumption that onEvent calls made from converters on their own timing may be the root cause of the error, but that is deep within code I cannot actively debug.

But somewhere this should be called: https://github.com/Koenkk/zigbee-herdsman-converters/blob/72a6b9db3420df1d69f844019b4a4a8dc7506a3b/src/index.ts#L568 , I think that call is not correctly awaited from iobroker

asgothian, 2025-01-10:
Thanks for the hint what to look for. I do have a question regarding async and error handling. I was working off the assumption that this would work ok:

```
async function a()
{
//   do something
  throw (error)
}

async function b() 
{
  await a();
}

async function c()
{
  try {
    await b();
  }
 catch () {}
}

```

is this the case, or would that already raise an uncaught exception ?

In any case - I will go through the code and check. I have to admit that I initially checked herdsman calls, not herdsman-converter calls.

A.

Koenkk, 2025-01-11:
That should work indeed.

asgothian, 2025-01-12:
Seems I caught most of the offenders. Closing this. Thanks again for the help.

closed 2025-01-12 by asgothian