Koenkk/zigbee-herdsman #1338: Use full paths for adapter lookup

lmcd, 2025-03-11:
Hi,

Thanks for the great library! I've successfully been able to integrate `zigbee-herdsman` into a native Swift application.
I've done this by creating a single-file executable using `pkg`, and a light wrapper that sends all events and logging to my app over a pipe.

The main issue in getting `zigbee-herdsman` building with `pkg`, was the relative module imports in `adapter.ts`. Dynamically passing the path of each adapter as a variable into `import` is confusing `pkg`.

Instead of mapping each adapter by name/relative path, would it be possible to instead switch over each adapter key with a separate `import` statement for each adapter, with the actual string literal passed into the import?

Thanks!

lmcd, 2025-03-11:
Also, the reason I'm not submitting this myself, as in my own example, I'm just manually hacking the generated Javascript in `node_modules`. I don't use TypeScript day-to-day and don't have a toolchain up and running.

Koenkk, 2025-03-12:
No problem, would you mind submitting a PR? Setting it up should be easy, just install node on your system, clone the repo, edit the code and run `npm run build && npm tst` to verify the changes.

Nerivec, 2025-03-15:
That should fix it, but looks like `pkg` may have an underlaying problem. The code was a bit "involved", but it should work.

lmcd, 2025-03-15:
This looks good - thanks for this.
In the meantime, I've decided to just modify zigbee2mqtt for my needs, but will still need to find a way of getting pkg to cooperate with all the dynamic imports - but switching to this more verbose method doesn't hurt at all.

closed 2025-03-16 by Koenkk