Koenkk/zigbee-herdsman #1353: Inconsistencies in add device with install code

JohnConnett, 2025-03-19:
These inconsistencies when adding a device using a 144-bit install code were identified in the discussion [Unsupported install code #19982](https://github.com/Koenkk/zigbee2mqtt/discussions/19982).

In [zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/adapter/zStackAdapter.ts) the MMO hash function is not used.

In [zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/ember/adapter/emberAdapter.ts) the MMO hash function is used _to prevent a compromise of the backup data from compromising the current link keys_ (This is not used for the ezsp or dconz adapters). While this is possibly a sensible security feature it may not be compatible with the [Open ZigBee Coordinator Backup Format](https://github.com/zigpy/open-coordinator-backup).

A concern is that the link-keys in backup files may not be consistent which could cause problems when restore is used with different adapters and systems.

Koenkk, 2025-03-20:
Could you submit a PR to fix this?

Nerivec, 2025-03-20:
I think topics are being mixed up here (only read the bottom of the discussion).

The hashing for the addition of an install code is made depending on what the stack expects (in case of zstack it is driven by a parameter, but it's only a matter of "doing it in Z2M" or "doing it in the firmware", I left a `todo` because it's more efficient to do it in Z2M, but previous codebase did not, and I couldn't test it, so, better to leave as is until someone can test it).
You do have to pass the install code _as the firmware expects it though_, and each stack has slight variations on that.

The hashing for backups, the code is there for `ember`, but `ember` doesn't actually use link keys (hence why there are no `devices` in `ember` backups, same for ZHA).

re: https://github.com/Koenkk/zigbee-herdsman/pull/1243
re: https://github.com/Koenkk/zigbee-herdsman/pull/1317

JohnConnett, 2025-03-20:
Thanks for the replies. I am trying to understand Zigbee security and the use of Install Codes and I am not confident that I know enough to submit a fix.

For some devices there is a 18 byte Install Code (16 byte code + 2 byte CRC). Typically, this is provided along with the 8 byte EUI-64 encoded into a QR Code, as separate hexadecimal strings, a NFC tag or other offline mechanism. The 18 byte Install Code is used as the input to the Matyas-Meyer-Oseas (MMO) hash function to generate a 16 byte value (Link Key?).

If I understand @Nerivec correctly, the Install Code can either be MMO hashed in software before passing it to the firmware or it can be passed directly to the firmware to apply the MMO hash internally. Either way, the firmware knows the MMO hash but will not have access to the original Install Code if the MMO hash is obtained in software.

I'm not sure what Link Key value should be stored in the backup for a device joining with an Install Code so that it could be successfully restored, including on systems with a different coordinator and stack. I thought it might be the MMO hash value but perhaps for zstack the tclk_seed is also involved?

Please feel free to correct any misunderstandings on my part.

Nerivec, 2025-03-20:
Chapter 10 in the first PDF (13-0402-13) explains this in more detail:
https://github.com/Koenkk/zigbee-herdsman/wiki/References

Also some of the tests show it in "limited" action:
https://github.com/Koenkk/zigbee-herdsman/blob/349cddfca7ff09e42e11d23f9270c48d44c735b3/test/zspec/utils.test.ts