Koenkk/zigbee-herdsman #1402: Supporting Inovelli install codes

kiler129, 2025-05-07:
## Problem
Every current device from Inovelli supports Install Codes, as per Zigbee 3.0 R23 spec. However, CSA never defined a format for these, so naturally Inovelli install codes (encoded in a QR on the device) triggers "Unsupported install code" message in Z2M.

## Inovelli format
I wasn't able to find any official specification from Inovelli regarding the format. However, looking at a handful of examples from `VZM31-SN` and `VZM35-SN`, the format appears to be `Z:<ieee>$I:<code>`, e.g. `Z:6C5CB1FFFE44FDFD$I:5492072F8DE72829FEE139CF8ACA4F43EF21`. This is similar to the current Aqara but with less metadata.

## Solution
Looking at the [`Controller.addInstallCode()`](https://github.com/Koenkk/zigbee-herdsman/blob/22057156f14bdebe7819e86945c3fc9ad5d28d28/src/controller/controller.ts#L248) this case can easily be handled with another regex match for Inovelli, or a more clever regex that handles both cases (`(?:^Z|\$A):(.+)\$I:(.+)$`) avoiding another `else if`.
Are any of these solutions acceptable?

Koenkk, 2025-05-08:
> avoiding another else if. Are any of these solutions acceptable?

Yes! Would you mind making a PR?

kiler129, 2025-05-08:
Sure, I will try to tackle it today or tomorrow :)

closed 2025-07-04 by Koenkk