Koenkk/zigbee-herdsman #1403: Suddenly configuration is not consistent with adapter state/backup

felixschndr, 2025-05-10:
After restarting my server I am hit with `Configuration is not consistent with adapter state/backup`. I did not change my config and all the values it tells my to compare are the same for my backup and my config:
```text
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	zh:adapter:zstack:manager: - PAN ID: configured=46573, adapter=46573
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	zh:adapter:zstack:manager: - Extended PAN ID: configured=b001fcc18fc7170f, adapter=b001fcc18fc7170f
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	zh:adapter:zstack:manager: - Network Key: configured=b0698ca8aa773d2b9cbe319c277a1e32, adapter=b0698ca8aa773d2b9cbe319c277a1e32
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	zh:adapter:zstack:manager: - Channel List: configured=11, adapter=11
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	zh:adapter:zstack:manager: Please update configuration to prevent further issues.
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	zh:adapter:zstack:manager: If you wish to re-commission your network, please remove coordinator backup at /app/data/coordinator_backup.json.
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	zh:adapter:zstack:manager: Re-commissioning your network will require re-pairing of all devices!
homeassistant_zigbee2mqtt  | [2025-05-10 20:20:24] error: 	z2m: Error while starting zigbee-herdsman
```
This issue was [also raised in the Zigbee2MQTT repo](https://github.com/Koenkk/zigbee2mqtt/issues/24103), however it belongs here.

The corresponding code is [here](https://github.com/Koenkk/zigbee-herdsman/blob/da33fbc5b72e5d5dbe2c11846d76e3b2c36f6116/src/adapter/z-stack/adapter/manager.ts#L142).

For now the only solution is to repair all devices which is a real bummer

felixschndr, 2025-05-10:
Debug log:
```text
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:05] info: 	z2m: Starting zigbee-herdsman (4.0.0)
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:05] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":46573,\"extendedPanID\":[176,1,252,193,143,199,23,15],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS10"},{"path":"/dev/ttyS11"},{"path":"/dev/ttyS12"},{"path":"/dev/ttyS13"},{"path":"/dev/ttyS14"},{"path":"/dev/ttyS15"},{"path":"/dev/ttyS16"},{"path":"/dev/ttyS17"},{"path":"/dev/ttyS18"},{"path":"/dev/ttyS19"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS20"},{"path":"/dev/ttyS21"},{"path":"/dev/ttyS22"},{"path":"/dev/ttyS23"},{"path":"/dev/ttyS24"},{"path":"/dev/ttyS25"},{"path":"/dev/ttyS26"},{"path":"/dev/ttyS27"},{"path":"/dev/ttyS28"},{"path":"/dev/ttyS29"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyS30"},{"path":"/dev/ttyS31"},{"path":"/dev/ttyS4"},{"path":"/dev/ttyS5"},{"path":"/dev/ttyS6"},{"path":"/dev/ttyS7"},{"path":"/dev/ttyS8"},{"path":"/dev/ttyS9"},{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"928f1faaca3aef118b45301455516304","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_928f1faaca3aef118b45301455516304-if00-port0","vendorId":"10c4","productId":"ea60"}]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"928f1faaca3aef118b45301455516304","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_928f1faaca3aef118b45301455516304-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 4
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":46573,"extendedPanID":[176,1,252,193,143,199,23,15],"channelList":[11]},"serialPort":{"baudRate":115200,"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] info: 	zh:zstack:znp: Serialport opened
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":1625}
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --- parseNext []
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":1625}
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --- parseNext []
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:znp: --> SREQ: SYS - version - {}
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,2,35]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: <-- [254,10,97,2,2,1,2,7,1,20,100,52,1,0,43]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --- parseNext [254,10,97,2,2,1,2,7,1,20,100,52,1,0,43]
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,20,100,52,1,0] - 43
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:znp: <-- SRSP: SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20210708}
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack:unpi:parser: --- parseNext []
homeassistant_zigbee2mqtt  | [2025-05-10 20:30:06] debug: 	zh:zstack: Adapter concurrent: 16
[... cut ...]
homeassistant_zigbee2mqtt  |     at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:154:16)
homeassistant_zigbee2mqtt  |     at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
homeassistant_zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:70:27)
homeassistant_zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:104:13)
homeassistant_zigbee2mqtt  |     at start (/app/index.js:149:5)
homeassistant_zigbee2mqtt exited with code 1
````

Koenkk, 2025-05-11:
I've added some additional logging, could you provide the debug log again?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

rupinski75, 2025-05-13:
Hello! I have similar problem but in my case after every restart backup does not match adapter configuration and commissioning process is started every time which means I have to readopt all devices again each restart.

I installed edge version with additional logging but my if branch of the [code](https://github.com/Koenkk/zigbee-herdsman/blob/da33fbc5b72e5d5dbe2c11846d76e3b2c36f6116/src/adapter/z-stack/adapter/manager.ts#L259C71-L259C87) does not have any logging at all :/

Is there anything I can do to troubleshoot the issue further?

Log:
<details>

```log [2025-05-13 10:06:47] info: 	z2m: Logging to console, file (filename: log.log)
[2025-05-13 10:06:47] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-05-13 10:06:47] info: 	z2m: Starting Zigbee2MQTT version 2.3.0-dev (commit #f656d962)
[2025-05-13 10:06:47] info: 	z2m: Starting zigbee-herdsman (4.0.2)
[2025-05-13 10:06:47] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":41164,\"extendedPanID\":[75,236,86,189,47,146,217,214],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":false,\"path\":\"/dev/ttyACM0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-05-13 10:06:48] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"Texas Instruments","serialNumber":"__0X00124B0014D9A90F","pnpId":"usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D9A90F-if00","vendorId":"0451","productId":"16a8"},{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"}]
[2025-05-13 10:06:48] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"Texas Instruments","serialNumber":"__0X00124B0014D9A90F","pnpId":"usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D9A90F-if00","vendorId":"0451","productId":"16a8"} => zstack: 4
[2025-05-13 10:06:48] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":41164,"extendedPanID":[75,236,86,189,47,146,217,214],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyACM0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-05-13 10:06:48] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-05-13 10:06:48] info: 	zh:zstack:znp: Serialport opened
[2025-05-13 10:06:48] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,121,1,26]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,121,1,26]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [121,1] - 26
[2025-05-13 10:06:48] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":377}
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-05-13 10:06:48] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,121,1,26]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,121,1,26]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [121,1] - 26
[2025-05-13 10:06:48] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":377}
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-05-13 10:06:48] debug: 	zh:zstack:znp: --> SREQ: SYS - version - {}
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,2,35]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: <-- [254,14,97,2,2,0,2,6,3,167,62,52,1,2,0,0,0,0,198]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --- parseNext [254,14,97,2,2,0,2,6,3,167,62,52,1,2,0,0,0,0,198]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,167,62,52,1,2,0,0,0,0] - 198
[2025-05-13 10:06:48] debug: 	zh:zstack:znp: <-- SRSP: SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20201127}
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-05-13 10:06:48] debug: 	zh:zstack: Adapter concurrent: 2
[2025-05-13 10:06:48] debug: 	zh:zstack: Detected znp version 'ZStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20201127})
[2025-05-13 10:06:48] debug: 	zh:adapter:zstack:manager: beginning znp startup
[2025-05-13 10:06:48] debug: 	zh:zstack:znp: --> SREQ: SYS - osalNvLength - {"id":130}
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:writer: --> frame [254,2,33,19,130,0,178]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,19,21,0,101]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,19,21,0,101]
[2025-05-13 10:06:48] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 19 - [21,0] - 101
[... cut ...]
[2025-05-13 10:06:56] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-05-13 10:06:56] debug: 	z2m: Zigbee network parameters: {"channel":11,"extendedPanID":"0x4bec56bd2f92d9d6","nwkUpdateID":0,"panID":41164}
[2025-05-13 10:06:56] info: 	z2m: Currently 0 devices are joined.
[2025-05-13 10:06:56] info: 	z2m: Connecting to MQTT server at mqtt://homeassistant:1883
[2025-05-13 10:06:56] debug: 	z2m: Using MQTT login with username: mqtt
[2025-05-13 10:06:56] info: 	z2m: Connected to MQTT server...
```
</details>

Koenkk, 2025-06-25:
@felixschndr could you provide the debug logging of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

closed 2025-09-04 by github-actions[bot]

felixschndr, 2025-10-15:
Per your [commit](https://github.com/CubeZ2mDeveloper/zigbee-herdsman/commit/a82b79fae602bfd74089cdbe1f917e8173b19795) you can see the log message `Network Key: configured=a21cef89d8efd560ab53ee5daf200ed6, adapter:active=a21cef89d8efd560ab53ee5daf200ed6 adapter:preconfigured=1acd5c006391d5142027942adfe8a210, adapter:alternate=a21cef89d8efd560ab53ee5daf200ed6`

felixschndr, 2025-10-15:
As I see it 
https://github.com/Koenkk/zigbee-herdsman/blob/7bad84cd8e4c21b4e60870c5a60f17a3ae559078/src/adapter/z-stack/adapter/manager.ts#L141
is false since the `networkKey` is not equal to the `preconfiguredKey`

Koenkk, 2025-10-16:
@felixschndr I mean reflashing the firmware on the stick ([docs](https://www.zigbee2mqtt.io/guide/adapters/zstack.html#firmware-flashing))

Koenkk, 2025-10-17:
@felixschndr by doing this you reset the adapter memory and upon starting z2m the backups get synced to it again. I think I'm going to disable this checks (as in very rare cases it seems to go wrong)

felixschndr, 2025-10-21:
> did you flash 20250321 now?

yes, I did. I've not had the issue since. However, as mentioned this issue only happened on hard shutdowns which I've not had in the past few days.

nicolas-lang, 2026-02-07:
I encountered this issue aswell, and only managed to fix it with a network reset.