Koenkk/zigbee-herdsman #1470: Problems using configure reporting

asgothian, 2025-08-02:
Hello,

i am trying to configure reporting on a number of devices, but am running into an odd error:

the code used is:

```
 async tryConfigureReporting(entity, cluster, attributes) {
        if (!entity.inputClusters.includes(cluster)) return;
        // Brute force - just try:
        const attr = [...attributes];
        const reportables = [];
        for (const attribute in attr) {
            reportables.push({
                attribute:attribute, 
                minimumReportInterval:0, 
                maximumReportInterval:65000,
                reportableChange:0})
        }
        try {
            await entity.configureReporting(cluster, reportables)
        } catch (error) {
            this.warn(`error on configure reporting: ${error && error.message ? error.message : 'no details given'}`);
        }

    }

    async triggerReporting(entity) {
        // check if the cluster is supported
        this.tryConfigureReporting(entity, 6,['onOff']);
        this.tryConfigureReporting(entity, 8,['currentLevel']);
        this.tryConfigureReporting(entity, 768,['currentHue','currentSaturation','currentX','currentY','colorTemp']);
    }
```

which generates an error, and does not complete the reporting:

```
2025-08-02 17:59:26.112 - warn: zigbee.0 (192015) error on configure reporting: ZCL command 0x0017880103124dc3/11 genOnOff.configReport([{"attribute":"0","minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Don't know value type for 'undefined')
2025-08-02 17:59:26.113 - warn: zigbee.0 (192015) error on configure reporting: ZCL command 0x0017880103124dc3/11 genLevelCtrl.configReport([{"attribute":"0","minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Don't know value type for 'undefined')
2025-08-02 17:59:26.113 - warn: zigbee.0 (192015) error on configure reporting: ZCL command 0x0017880103124dc3/11 lightingColorCtrl.configReport([{"attribute":"0","minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":0},{"attribute":"1","minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":0},{"attribute":"2","minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":0},{"attribute":"3","minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":0},{"attribute":"4","minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Don't know value type for 'undefined')

```

I have tried using the cluster ID's instead of the cluster names, but this generates the same issue, which is raised deep within the herdsman. What confuses me is that the same code seems to be used from within the zigbee-herdsman-converters for a number of devices, and there is no 'undefined' passed in my call.

I am using the Herdsman version 5.0.3.

Any hint as to what the cause could be, and how to mitigate it ?

thank you for the help.

asgothian, 2025-08-02:
I did run a test with ZH 4.5.0 - with the same result

Nerivec, 2025-08-11:
You can easily run pieces of code like this (plain JS, no nodejs) in an online exec, or even any browser's console.
```js
        const attributes = ['currentHue','currentSaturation','currentX','currentY','colorTemperature'];
        const attr = [...attributes];
        const reportables = [];
        for (const attribute in attr) {
            reportables.push({
                attribute:attribute, 
                minimumReportInterval:0, 
                maximumReportInterval:65000,
                reportableChange:0})
        }
        console.log(reportables);
```

[image]

closed 2025-08-11 by Nerivec