Koenkk/zigbee-herdsman #1498: `UINT8` doesn't allow `0xff` (255) as value anymore which breaks code

DerDreschner, 2025-09-07:
With in the latest releases of Z2M, there must have been a change in the code regarding processing of messages with `UINT8`. This leads to the situation, that 255 isn't a valid value anymore, although it is valid, and it's been used multiple times ([see for example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/converters/toZigbee.ts#L769)).

I realized that when I implemented [this pull request](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9941), I didn't get any value regarding the light settings as well as the `maximumBrightness` attribute, which allows 255 as a valid value as well (it's actually the default value). The `meta` attribute contains the correct byte values in that case, but I'm not sure if that was after some changes in `zigbee-herdsman` or not. I'm not the only person who realized the issue. See https://github.com/Koenkk/zigbee2mqtt/issues/28506 for example.

Is this a bug in `zigbee-herdsman` or do all converters that need to process `0xff` as a valid value have to be changed to the `raw` message type?

DerDreschner, 2025-09-07:
That's correct. Setting `0xff` does work correctly. Just reading it is broken.

Nerivec, 2025-09-08:
I dug a little more. There is the base level control, and a "level control for lighting". The latter is described in the Zigbee spec at 3.19, page 307. Both have ID 0x0008, the "for lightning" is "derived". Use of derived clusters is recommended over the generic base.
The base level control cluster has a device-defined range, that defaults to 0x00..0xff range if not specified, with a default of 0xff (it's not indicated as non-value, so I assume this is a "plain" value, as the range is clearly different otherwise - per below).
The level control for lighting cluster has a fixed 0x01..0xfe range, with a default of 0xff that is used to indicate undefined (non-value).

I took a quick look at the Silabs code, it confirms all the above behaviors.

The current ZH result should be correct for most cases (although zero is also prohibited which is not technically handled...), but we'd need a way to differentiate for non-lighting use. However, this is the only case of "derived" I could find in the spec, so, maybe the base just is expected never to be used (assumed to be a backwards-compat scheme from early spec days)? Not sure how we fit that extra handling otherwise since IDs are identical. _The PWM cluster (3.20) also derives from the same base level control, but that one has a different ID._

It also appears we do need to handle the specific "default"s in tables as plain values, and not consider them non-values when present and matching the non-values for the types. While the current ZH result ends up correct for this specific issue, the processing is not technically accurate (likely mismatches for other clusters). We need a way to specify these defaults and to ignore non-values as appropriate, something like this for all types:
```diff
-        return value === 0xff ? Number.NaN : value;
+        return value === 0xff && valDefault !== value ? Number.NaN : value;
```